carla-simulator/scenario_runner #18: How to find and set the starting position while making custom scenarios in carla 0.9.2?

anshulpaigwar, 2018-12-27:
Hello, I am trying to make my own custom scenario similar to no_signal_junction_crossing but I want a different junction hence to start positions would be different of the vehicles for the new junction scenario.
But I am having difficulties finding the coordinates. As I could not find any mapping between carla world in Euclidean space.

pravinblaze, 2018-12-27:
Hi Anush, checkout this [tool](https://github.com/pravinblaze/carla-code-pad/blob/master/displaytopology.py). You can read the co-ordinates from the matplotlib plot. Opendrive and Unreal use co-ordinate systems of different chirality so just multiply your x values from the plot by -1. Y values are fine. The script just plots end points of road segments so you cant see any curves. This [script](https://github.com/pravinblaze/carla-code-pad/blob/master/opendrive.py) extracts an opendrive string for the map. You can visualise it with [opendrive viewer](http://www.opendrive.org/tools/odrViewer64.zip) .

anshulpaigwar, 2018-12-27:
Thanks @pravinblaze it solved my problem.

closed 2018-12-27 by anshulpaigwar