carla-simulator/scenario_runner #19: AttributeError: module 'carla' has no attribute 'Transform'

chowkamlee81, 2019-01-05:
When trying to execute python scenario_runner.py --scenario FollowLeadingVehicle
got the error below

Traceback (most recent call last):
  File "scenario_runner.py", line 22, in <module>
    from Scenarios.follow_leading_vehicle import *
  File "/home/deepaknayak/Documents/Reinforcement-learning/scenario_runner-master/Scenarios/follow_leading_vehicle.py", line 30, in <module>
    class FollowLeadingVehicle(BasicScenario):
  File "/home/deepaknayak/Documents/Reinforcement-learning/scenario_runner-master/Scenarios/follow_leading_vehicle.py", line 44, in FollowLeadingVehicle
    _ego_vehicle_start = carla.Transform(
AttributeError: module 'carla' has no attribute 'Transform'

fabianoboril, 2019-01-06:
Hi,
did you setup CARLA as described in the documentation, e.g. did you properly set the PYTHONPATH?
https://github.com/carla-simulator/scenario_runner/blob/master/Docs/getting_started.md#running-the-follow-vehicle-example

fabianoboril, 2019-01-07:
Hi,

when you just start python, which version does it print? 3.5 or 2.x? If it is 2.x, please update your PYTHONPATH to use the correct egg-file, or start the scenario with python3.

fabianoboril, 2019-01-11:
Any updates on this one? Is it resolved?

geekhut, 2019-01-17:
Try with python3:

`python3 scenario_runner.py --scenario FollowLeadingVehicle
`

fabianoboril, 2019-01-23:
I guess the problem is solved. I'll close the issue for now.

closed 2019-01-23 by fabianoboril