carla-simulator/scenario_runner #23: AttributeError: type object 'ParallelPolicy' has no attribute 'SUCCESS_ON_ONE'

rguptatx, 2019-01-22:
When I run scenario runner on ubuntu 16.04 with python3 I get this or similar error with all scenarios:
Traceback (most recent call last):
  File "scenario_runner.py", line 160, in <module>
    main(ARGUMENTS)
  File "scenario_runner.py", line 97, in main
    scenario = scenario_class(world, args.debug)
  File "/usr/local/scenario_runner/Scenarios/object_crash_vehicle.py", line 152, in __init__
    debug_mode=debug_mode)
  File "/usr/local/scenario_runner/Scenarios/basic_scenario.py", line 79, in __init__
    behavior = self._create_behavior()
  File "/usr/local/scenario_runner/Scenarios/object_crash_vehicle.py", line 191, in _create_behavior
    policy=py_trees.common.ParallelPolicy.SUCCESS_ON_ONE)
AttributeError: type object 'ParallelPolicy' has no attribute 'SUCCESS_ON_ONE'

fabianoboril, 2019-01-22:
Hi,
which version of py_trees do you use? We support py_trees 0.8.x. It seems things have changed with version 1.x (there it is SuccessOnOne).

closed 2019-01-22 by fabianoboril

GeneralMak, 2021-07-23:
@rguptatx  hello,Have you created a successful scenario by referring to the example of the official scenario creation?(https://carla-scenariorunner.readthedocs.io/en/latest/creating_new_scenario.html)