carla-simulator/scenario_runner #24: Support py_trees 1.x

fabianoboril, 2019-01-22:
Currently we support only py_trees 0.8.x. Now there is a new release, 1.x:

https://py-trees.readthedocs.io/en/release-1.0.x/index.html

However, with the latest py_trees release, several namings have changed, for example SUCCESS_ON_ONE became SuccessOnOne. We should update to the latest py_trees release and adapt our namings.

fabianoboril, 2019-01-23:
The changelog can be found here:
https://py-trees.readthedocs.io/en/release-1.0.x/changelog.html#changelog

Changes that are problematic for us:
SUCCESS_ON_ONE --> SuccessOnOne()
SUCCESS_ON_ALL --> SuccessOnAll()
No more Python2 support

fabianoboril, 2019-01-23:
See also PR #26

germanros1987, 2019-01-25:
Is our update intended to benefit from these new features?: 

[composites] new parallel policies, SuccessOnAll, SuccessOnSelected
[decorators] oneshot policies for activating on completion or successful completion only

What is the good stuff of 1.0 that is making us update?

fabianoboril, 2019-01-25:
The new parallel policies may be useful. At the same time, you could mimic this behavior also with py_trees 0.8.

julienlau, 2021-10-08:
FYI
Please find below a poetry configuration that seems to work on ubuntu20.04:
```
[jlu@linux-gp75]> cat ~/carla-test/pyproject.toml
[tool.poetry]
name = "carla-test"
version = "0.0.1"
description = "test carla"
authors = ["Julien Laurenceau <julien.laurenceau@pepitedata.com>"]

[tool.poetry.dependencies]
python = ">=3.7,<3.8"
xmlschema = "^1.8.0"
numpy = "^1.21.2"
six = "^1.16.0"
#py-trees = "^2.1.6"
py-trees = "<0.9"
ephem = "^4.1"
networkx = "^2.6.3"
Shapely = "^1.7.1"
tabulate = "^0.8.9"

[tool.poetry.dev-dependencies]

[build-system]
requires = ["poetry-core>=1.0.0"]
build-backend = "poetry.core.masonry.api"
```