carla-simulator/scenario_runner #35: ImportError: No module named agents.navigation.roaming_agent

GetDarren, 2019-02-04:
OS: Ubuntu 18.04
Python: 2.7
I'm trying to run the :  python scenario_runner.py

```

Traceback (most recent call last):
  File "scenario_runner.py", line 25, in <module>
    from Scenarios.follow_leading_vehicle import *
  File "/home/cienet/scenario_runner/Scenarios/follow_leading_vehicle.py", line 24, in <module>
    from ScenarioManager.atomic_scenario_behavior import *
  File "/home/cienet/scenario_runner/ScenarioManager/atomic_scenario_behavior.py", line 20, in <module>
    from agents.navigation.roaming_agent import *
ImportError: No module named agents.navigation.roaming_agent
```

Here is my pip list:
```
Package       Version
------------- -------
asn1crypto    0.24.0 
carla         0.9.2  
coverage      4.5.2  
cryptography  2.1.4  
decorator     4.1.2  
enum34        1.1.6  
funcsigs      1.0.2  
idna          2.6    
ipaddress     1.0.17 
keyring       10.6.0 
keyrings.alt  3.0    
mock          2.0.0  
networkx      1.11   
nose2         0.8.0  
numpy         1.13.3 
pbr           5.1.2  
pip           19.0.1 
pkg-resources 0.0.0  
py-trees      0.8.3  
pycrypto      2.6.1  
pydot         1.4.1  
pygame        1.9.4  
pygobject     3.26.1 
pyparsing     2.3.1  
pyxdg         0.25   
PyYAML        3.12   
SecretStorage 2.3.1  
setuptools    40.7.3 
six           1.12.0 
wheel         0.32.3 
```

Please kindly help me out,  if I missed to install any dependencies?
Thank you .

Soolek, 2019-02-05:
I have the same problem after upgrading to latest master branch.

Removing the lines 20,21 from atomic_scenario_behavior did fix the issue:
`from agents.navigation.roaming_agent import *
from agents.navigation.basic_agent import *`

GetDarren, 2019-02-05:
> I have the same problem after upgrading to latest master branch.
> 
> Removing the lines 20,21 from atomic_scenario_behavior did fix the issue:
> from agents.navigation.roaming_agent import *
> from agents.navigation.basic_agent import *

Thank you , I tried to change the PYTHONPATH and it also works.

```
CARLA_ROOT=/home/cienet/CARLA_0.9.2
export PYTHONPATH=$PYTHONPATH:${CARLA_ROOT}/PythonAPI/carla-0.9.2-py3.5-linux-x86_64.egg:${CARLA_ROOT}/PythonAPI
```

closed 2019-02-05 by GetDarren

violayu5201, 2023-05-22:
I have configured the environment variables, but Python also throw the error:
      ModuleNotFoundError: No module named 'agents.navigation.constant_velocity_agent'

endris1011, 2023-05-30:
> I have configured the environment variables, but Python also throw the error: ModuleNotFoundError: No module named 'agents.navigation.constant_velocity_agent'

I have currently the same problem, do you have a solution yet? @violayu5201

Sid-B-K, 2025-12-24:
> I have configured the environment variables, but Python also throw the error: ModuleNotFoundError: No module named 'agents.navigation.constant_velocity_agent'

Just copy paste this file in the agents folder - https://github.com/carla-simulator/carla/blob/e41ffa03c27e381f52bc98402cc3100aac225ac2/PythonAPI/carla/agents/navigation/constant_velocity_agent.py