carla-simulator/scenario_runner #37: Carla [0.9.2] getting slower after each repetition of scenario_runner

anshulpaigwar, 2019-02-06:
In scenario_runner --version 0.1. I found that the FPS of Carla Server drops after every repetition. I think the problem being that actors are not properly destroyed from the server. After the end of scenario I get this output:

```sh
LIFETIME: WARNING! the following objects were not destructed. 
LIFETIME: Actor 168 (sensor.other.collision) still alive. 
LIFETIME: tcp client 94 still alive. 
LIFETIME: Actor 163 (vehicle.tesla.model3) still alive. 
LIFETIME: tcp client 89 still alive. 
LIFETIME: tcp client 88 still alive. 
LIFETIME: Actor 159 (sensor.other.collision) still alive. 
LIFETIME: Actor 155 (vehicle.tesla.model3) still alive. 
LIFETIME: Actor 152 (sensor.other.collision) still alive. 
LIFETIME: tcp client 82 still alive. 
LIFETIME: Actor 151 (sensor.other.collision) still alive. 
LIFETIME: Actor 148 (vehicle.nissan.micra) still alive. 
LIFETIME: Actor 147 (vehicle.tesla.model3) still alive. 
LIFETIME: tcp client 77 still alive. 
LIFETIME: Actor 144 (sensor.other.collision) still alive. 
LIFETIME: tcp client 76 still alive. 
LIFETIME: Actor 143 (sensor.other.collision) still alive. 
LIFETIME: Actor 167 (sensor.other.collision) still alive. 
LIFETIME: Actor 139 (vehicle.tesla.model3) still alive. 
LIFETIME: tcp client 28 still alive. 
LIFETIME: Actor 164 (vehicle.nissan.micra) still alive. 
LIFETIME: Actor 120 (sensor.other.collision) still alive. 
LIFETIME: Actor 55 (sensor.other.collision) still alive. 
LIFETIME: Actor 156 (vehicle.nissan.micra) still alive. 
LIFETIME: Actor 136 (sensor.other.collision) still alive. 
LIFETIME: tcp client 23 still alive.
```
I am doing `actor.destroy` and `actor = None` after each repetition and they are being destroyed (I cannot see them in the server) but the memory still not beig released and I get above output. 

Could someone help me how to destroy the actors properly??

Danmou, 2019-02-28:
I'm seeing the same problem.

anshulpaigwar, 2019-03-06:
@Danmou you are correct this is happening with me too.

fabianoboril, 2019-03-08:
@anshulpaigwar @Danmou Does this problem still exist when you use CARLA 0.9.4 and the latest version from the development branch?

Could you please provide the exact setup that you use? CARLA version, how you start CARLA, what kind of client do you run etc. Thank you.

damascenodiego, 2019-05-22:
This is happening to me as well with Carla version 0.9.5

TigerStone93, 2022-11-01:
Is this problem solved?