carla-simulator/scenario_runner #40: How do I get the exact location (x, y, z) on the map to spawn actors?

cienet1, 2019-02-12:
Hello,
I'm new to Carla.
Got a question while I was just trying out the API by creating some sample scenarios. 
I was wondering if there is an easy way to find out  the location value of a point on the map.
Thanks for your time!

fabianoboril, 2019-02-25:
Either you open the map in the Unreal editor and find the desired coordinates, or you drive with the manual client to the position you like, and note then note the coordinates.

damascenodiego, 2019-04-25:
Recently, I've implemented a small [code to print the spectactor's current position on Carla 0.9.5.](https://gist.github.com/damascenodiego/c53ae6960f1ebdcc25d41384392b6500)

Remind that the scenario runner's release works for 0.9.2 and you will need to make some adjusts on it. More specifically, you'll need to **change line 14** to the right path where the .egg files are found.

ornare, 2019-07-17:
@fabianoboril Is the z coordinate data stored in OpenDrive files maybe?

closed 2020-03-02 by fabianoboril

youngsend, 2021-03-02:
I opened the OpenDRIVE file with `odrViewer`. `odrViewer` will show x, y, z coordinate of the cursor position. However the free `odrViewer` seems to have disappeared from Internet...

henkenen, 2022-01-19:
> Either you open the map in the Unreal editor and find the desired coordinates, or you drive with the manual client to the position you like, and note then note the coordinates.

Could you give me some hints on how to get the coordinates in Unreal editor?