carla-simulator/scenario_runner #48: ImportErro when run python scenario_runner.py

dyli96, 2019-02-22:
Hi, I want to use scenario_runner on carla.
I first export PYTHONPATH=$PYTHONPATH:$~/carla/PythonAPI/dist/carla-0.9.3.egg
Then python scenario_runner.py --scenario FollowLeadingVehicle
I always encounter the error:
 import carla
ImportError: No module named carla

I would really appreciate if anyone can help me with it, thanks in advance!

feipan664, 2019-03-19:
> It is fixed now. Then I encounter new errors.
> 
> del SCENARIORUNNER
> NameError: name 'SCENARIORUNNER' is not defined

You may have to run both CARLA and python command at the same time.

fandaolong, 2019-04-14:
I still met this issue in windows, anyone can give some hint on it.

fandaolong, 2019-04-14:
@dyli96, would you like share how did you fix the issue, I met the same problem now.

kraken24, 2019-05-27:
I import carla correctly but I still get this error...
Can anyone help??

Traceback (most recent call last):
  File "scenario_runner.py", line 30, in <module>
    from srunner.scenarios.opposite_vehicle_taking_priority import *
  File "/home/z661107/001-Repositories/scenario_runner/srunner/scenarios/opposite_vehicle_taking_priority.py", line 30, in <module>
    class OppositeVehicleRunningRedLight(BasicScenario):
  File "/home/z661107/001-Repositories/scenario_runner/srunner/scenarios/opposite_vehicle_taking_priority.py", line 48, in OppositeVehicleRunningRedLight
    _ego_end_position = carla.Location(x=-3, y=-90, z=0)   # End position
AttributeError: module 'carla' has no attribute 'Location'

glopezdiest, 2021-11-23:
Make sure your PYTHONPATH is correctly pointing to the *.egg* file

closed 2021-11-23 by glopezdiest