carla-simulator/scenario_runner #70: NameError: name 'challenge_evaluator' is not defined

overlapjho, 2019-03-05:
I installed successfully the Challenge Evaluator using `./setup_environment.sh --carla-root <path_to_the_CARLA_root_folder>` and test `python srunner/challenge/challenge_evaluator.py --list` which gives me:
```
Currently the following scenarios are supported:
NoSignalJunctionCrossing
Challenge_Basic_00
Challenge_Basic_01
Challenge_Basic_03
Challenge_Basic_04
StationaryObjectCrossing
DynamicObjectCrossing
FollowLeadingVehicle
FollowLeadingVehicleWithObstacle
ControlLoss
OppositeVehicleRunningRedLight
VehicleTurningRight
VehicleTurningLeft
```

trying to run with python3 gives me error such as : 
```
Traceback (most recent call last):
  File "srunner/challenge/challenge_evaluator.py", line 23, in <module>
    import carla
  File "/home/overlapjho/carla/PythonAPI/carla-0.9.3-py2.7-linux-x86_64.egg/carla/__init__.py", line 7, in <module>
  File "/home/overlapjho/carla/PythonAPI/carla-0.9.3-py2.7-linux-x86_64.egg/carla/libcarla.py", line 7, in <module>
  File "/home/overlapjho/carla/PythonAPI/carla-0.9.3-py2.7-linux-x86_64.egg/carla/libcarla.py", line 6, in __bootstrap__
  File "/usr/lib/python3.5/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: /home/overlapjho/.python-eggs/carla-0.9.3-py2.7-linux-x86_64.egg-tmp/carla/libcarla.so: undefined symbol: PyString_Type
```

So when I execute `bash srunner/challenge/run_evaluator.sh` or `python srunner/challenge/challenge_evaluator.py --scenario group:ChallengeBasic -a srunner/challenge/autoagents/DummyAgent.py` it throws me this error: 
```Traceback (most recent call last):
  File "srunner/challenge/challenge_evaluator.py", line 509, in <module>
    del challenge_evaluator
NameError: name 'challenge_evaluator' is not defined```
Help please

ivashmak, 2019-03-07:
same error

ivashmak, 2019-03-07:
Try to run with python3

JustWon, 2019-03-07:
I tried to run the script with python3.
the scenario window is working well but the error "name basestring is not defined" occured.

micmarty, 2019-03-07:
@ivashmak @overlapjho 
I had the same error and this worked for me:
`pip install opencv-python numpy pygame`

Let me know if it helped :)

overlapjho, 2019-03-07:
@micmarty didn't work because your issue involves missing scipy, while we get this ` undefined symbol: PyString_Type` problem

micmarty, 2019-03-08:
Yes, you're right, sorry :sweat_smile: It can still help others, though (according to common issue title).

stevendae, 2019-03-08:
`python3 srunner/challenge/challenge_evaluator.py --scenario group:ChallengeBasic -a srunner/challenge/autoagents/DummyAgent.py` works when you run it with python3 and you remove the 2.7 version egg in the Python API file.

However when I run the manual agent examples like:
`bash srunner/challenge/run_evaluator.sh`
or 
`python3 challenge_evaluator.py --scenario config:ChallengeBasic -a srunner/challenge/autoagents/HumanAgent.py --route-visible`

I always get the error:
AttributeError: 'ChallengeEvaluator' object has no attribute '_sensors_list'

Anyone fix this issue?

stevendae, 2019-03-08:
I solved my problem. The problem lies in this item. Although the python3 path also resides in sys.path, the python2.7 path precedes it. Python will catch the one in python2.7 first. When you are running in directly inside /python3/dist-packages, the current directory is placed first, and that precedes python2.

Solution:
Remove python2.7/dist-packages from your PYTHONPATH

overlapjho, 2019-03-09:
@stevendae  I tried to remove  python2.7/dist-packages from my PYTHONPATH, the command still reads carla-0.9.3-py2.7-linux-x86_64.egg and still suffer same issue, any idea how to solve this?

togaen, 2019-03-27:
I am also seeing this error. Does scenario runner require python3? If so, the documentation needs to be updated.

Johnnyswong, 2019-04-17:
Try checkout the carla_challenge branch to the master branch...

aadityasp1, 2019-05-31:
> Try to run with python3

I have tried with python2 it gave me the same errors as above,so tried with python3 (added the python3.5  .egg file to my PYTHONPATH as well) but it throws the following errors:

**$ python3 challenge_evaluator.py --scenario NoSignalJunctionCrossing --debug

Traceback (most recent call last):
  File "challenge_evaluator.py", line 28, in <module>
    from srunner.scenarios.challenge_basic import *
  File "/mnt/data2/scenario_runner-master/srunner/scenarios/challenge_basic.py", line 11, in <module>
    import py_trees
  File "/usr/local/lib/python3.5/dist-packages/py_trees/__init__.py", line 22, in <module>
    from . import decorators  # noqa
  File "/usr/local/lib/python3.5/dist-packages/py_trees/decorators.py", line 197
    ):
    ^
SyntaxError: invalid syntax**

I have updated my pytrees module as well on pip3 but it still gives the above error.

Anyone please help me fix this error..!

closed 2019-06-09 by jmarrr

havefun28, 2019-07-03:
Hi! May I ask where you guys get the file `challenge_evaluator.py`? I only find `challenge_evaluator_routes.py` in the folder `srunner/challenge`

deepthi53, 2020-04-11:
Hi Guys pls solve the below error:
NameError: name 'NUM_FEATURES' is not defined
feature_set = []
for idx in range(NUM_FEATURES):
    feature_unit = Lambda(lambda x : x[:,:,:,idx], output_shape=input_shape[1:3])(image_feature)
    feature_set.append(feature_unit)
    
# combine feature x feature into a set of relations
relation_set = []
for f_i in feature_set:
    for f_j in feature_set:
        if (f_i != f_j):
            relation_unit = keras.layers.concatenate([f_i, f_j])
            relation_set.append(relation_unit)

---------------------------------------------------------------------------
NameError                                 Traceback (most recent call last)
<ipython-input-10-84fd37bb07a2> in <module>()
      1 feature_set = []
----> 2 for idx in range(NUM_FEATURES):
      3     feature_unit = Lambda(lambda x : x[:,:,:,idx], output_shape=input_shape[1:3])(image_feature)
      4     feature_set.append(feature_unit)
      5 

NameError: name 'NUM_FEATURES' is not defined