carla-simulator/scenario_runner #81: import agent error

zlw21gxy, 2019-03-13:
ModuleNotFoundError: No module named 'agents' 
and I already try
export PYTHONPATH=$PYTHONPATH:~/Documents/carla94/PythonAPI/carla-0.9.4-py3.5-linux-x86_64.egg:~/Documents/PythonAPI   
please help, thanks

kk2491, 2019-03-14:
Can you please paste the complete error trace ? It would be easy to check if we know where exactly you are facing the issue.

eloy96vg, 2019-03-26:
Hi, i have a similar error. Attached screenshot.
[image]

Please help, thanks.

kk2491, 2019-03-26:
Export the egg file location to pythonpath

monjo44, 2019-04-01:
This might help you! At the bottom!
https://carla.readthedocs.io/en/latest/build_system/

hangqiu, 2019-05-29:
The egg file is in my pythonpath, but still the same error, my carla version is 0.9.5

Traceback (most recent call last):
  File "scenario_runner.py", line 26, in <module>
    from srunner.scenarios.follow_leading_vehicle import *
  File "/home/hang/research/AVRCarla/scenario_runner/srunner/scenarios/follow_leading_vehicle.py", line 24, in <module>
    from srunner.scenariomanager.atomic_scenario_behavior import *
  File "/home/hang/research/AVRCarla/scenario_runner/srunner/scenariomanager/atomic_scenario_behavior.py", line 20, in <module>
    from agents.navigation.roaming_agent import *
ImportError: No module named 'agents'

kk2491, 2019-05-29:
You need to update the sys.path for this

hangqiu, 2019-05-30:
sys.path does include the egg file :-) not sure what's happening here

hangqiu, 2019-05-30:
It turns out my "agents" folder is under PythonAPI/carla... adding that to pythonpath worked. Thanks!

closed 2019-06-28 by fabianoboril

raeed17, 2020-02-06:
@eloy96vg I'm having the same issue, how did you resolve this for the scenario runner? I did everything in the getting started doc, but getting that import error.

anna-srl, 2020-03-30:
Hi all and thank you for your amazing work!

I have been having the same issue for a while, I'm trying to use scenario runner on windows 10 (Carla 0.95) and I keep getting the error "no module named agents".

I have already created the system environment variables called CARLA_ROOT=C:\...\CARLA and PYTHONPATH=C:\..\CARLA\PythonAPI. I also tried to include the agents as described above by changing the variable to PYTHONPATH=C:\..\CARLA\PythonAPI\carla\agents, but still I get the same error. @hangqiu could you please explain exaclty the process you followed? 

Any suggestions are more than welcome, many thanks in advance :)

anna-srl, 2020-04-08:
I fixed it by adding the environment variables (windows 10, carla 0.95, Python 3.7):

CARLA_ROOT= C:...\CARLA
PYTHONPATH= C:...\Python ; C:...\CARLA\PythonAPI\carla; C:\...\CARLA\PythonAPI\carla\dist\carla-0.9.5-py3.7-win-amd64.egg ; C:\...\CARLA\PythonAPI\scenario_runner-0.9.5; C:\...\CARLA\PythonAPI\carla\agents

And added both CARLA_ROOT and PYTHONPATH to the Path system variable.
(there is a chance not all of the above paths were required in PYTHONPATH but after several trials this was proven to work)

sindujanarra, 2021-02-22:
from agents.navigation.roaming_agent import RoamingAgent
ImportError: No module named agents.navigation.roaming_agent
[Finished in 1.1s with exit code 1]
even after adding both CARLA_ROOT and PYTHONPATH to the Path system variable. it still throws error

alibukharai, 2021-10-09:
> Hi, i have a similar error. Attached screenshot. [image]
> 
> Please help, thanks.

Hi, I am wondering if you solve the issue, Can you help me?

alibukharai, 2021-10-09:
> I fixed it by adding the environment variables (windows 10, carla 0.95, Python 3.7):
>
> CARLA_ROOT= C:...\CARLA PYTHONPATH= C:...\Python ; C:...\CARLA\PythonAPI\carla; [...]

Hi I am wondering if you solve the issue, Can you help me?

HakanAkkurt, 2021-11-30:
> 

Has anyone any solution ? I have the same error

HakanAkkurt, 2021-11-30:
> > 
> 
> Has anyone any solution ? I have the same error

> Has anyone any solution ? I have the same error

HakanAkkurt, 2021-11-30:
> from agents.navigation.roaming_agent import RoamingAgent
ImportError: No module named agents.navigation.roaming_agent
[Finished in 1.1s with exit code 1]
even after adding both CARLA_ROOT and PYTHONPATH to the Path system variable. it still throws error

Has anyone any solution ? I have the same error...

Lucifer2700, 2022-01-09:
For Windows users I resolved the issue with running the command

set CARLA_ROOT= C:...\CARLA
set PYTHONPATH= C:...\Python ; C:...\CARLA\PythonAPI\carla; C:...\CARLA\PythonAPI\carla\dist\carla-0.9.5-py3.7-win-amd64.egg ; C:...\CARLA\PythonAPI\scenario_runner-0.9.5; C:...\CARLA\PythonAPI\carla\agents

vignesharu, 2022-08-24:
> For Windows users I resolved the issue with running the command
> 
> set CARLA_ROOT= C:...\CARLA set PYTHONPATH= C:...\Python ; C:...\CARLA\PythonAPI\carla; C:...\CARLA\PythonAPI\carla\dist\carla-0.9.5-py3.7-win-amd64.egg ; C:...\CARLA\PythonAPI\scenario_runner-0.9.5; C:...\CARLA\PythonAPI\carla\agents

After doing this, restart your command window.
The change in environmental variables takes place only after the restart of the command window.
[image]

SExpert12, 2023-06-17:
Ubuntu: 20.04(focal)
Carla: 0.9.14
Set path as follow:
export CARLA_ROOT=/home/ryzen/Downloads/CARLA14
export CARLA_ROOT=/home/ryzen/Downloads/CARLA14
export PYTHONPATH=$/usr/bin/python3:$/home/ryzen/Downloads/CARLA14/PythonAPI/carla/dist/carla-0.9.14.-py3.7-linux-x86_64.egg
export PYTHONPATH=$/usr/bin/python3:$/home/ryzen/Downloads/CARLA14/PythonAPI/carla
[image]
Facing this same error: No module named 'agents'