carla-simulator/scenario_runner #87: Spawning Problem for ts02 in town 02.

zaaramariyam, 2019-03-18:
On running this line: 
spawn_location, _ = get_location_in_distance(ego_vehicle, 20).
We get different locations at different times in Town 02. 

For example:
1. Starting location: 
<ego_vehicle x="27" y="110" z="0.4" yaw="180" model="vehicle.lincoln.mkz2017" />
Result:
Sometimes we get: (-7.768947124481201, 118.66976165771484, 1.0)
Sometimes we get: (48.00009536743164, 109.64213562011719, 1.0)

2. Starting location: 
<ego_vehicle x="28.7" y="302.5" z="0.4" yaw="180" model="vehicle.lincoln.mkz2017" />
Sometimes we get: (7.699801445007324, 302.9414978027344, 1.0)
Sometimes we get: (-7.768947124481201, 118.66976165771484, 1.0)

Function:
def get_location_in_distance(actor, distance):
    """
    Obtain a location in a given distance from the current actor's location.
    Note: Search is stopped on first intersection.

    @return obtained location and the traveled distance
    """
    waypoint = actor.get_world().get_map().get_waypoint(actor.get_location())
    traveled_distance = 0
    while not waypoint.is_intersection and traveled_distance < distance:
        waypoint_new = waypoint.next(1.0)[-1]
        traveled_distance += waypoint_new.transform.location.distance(waypoint.transform.location)
        waypoint = waypoint_new

    return waypoint.transform.location, traveled_distance

fabianoboril, 2019-03-19:
Hi,
can you check the cited function where exactly the difference occurs? Is it the first waypoint or on calling next()?

zaaramariyam, 2019-03-21:
The problem here is with the z-axis.
In town-02, wherever we spawn the actor it is showing"Unable to spawn actor due to collision at spawn point". 
And the cited function always returns z-value=0.

zaaramariyam, 2019-03-21:
def _initialize_actors(self, config):
        """
        Custom initialization
        """

        first_vehicle_waypoint, _ = get_waypoint_in_distance(self._reference_waypoint, self._first_vehicle_location)
        second_vehicle_waypoint, _ = get_waypoint_in_distance(self._reference_waypoint, self._second_vehicle_location)
        rotation = first_vehicle_waypoint.transform.rotation
        first_vehicle_spawn_location = carla.Transform(carla.Location(x=first_vehicle_waypoint.transform.location.x,
                                                                y=first_vehicle_waypoint.transform.location.y,
                                                                z=first_vehicle_waypoint.transform.location.z+1),
                                                                rotation)
        yaw_1 = second_vehicle_waypoint.transform.rotation.yaw + 90
        second_vehicle_spawn_location = carla.Transform(carla.Location(x=second_vehicle_waypoint.transform.location.x,
                                                                y=second_vehicle_waypoint.transform.location.y,
                                                                z=second_vehicle_waypoint.transform.location.z+1),
                                                        carla.Rotation(
                                                       pitch=second_vehicle_waypoint.transform.rotation.pitch,
                                                       yaw=yaw_1, roll=second_vehicle_waypoint.transform.rotation.roll))
        first_vehicle = CarlaActorPool.request_new_actor('vehicle.nissan.patrol', first_vehicle_spawn_location)
        second_vehicle = CarlaActorPool.request_new_actor('vehicle.diamondback.century',
                                                          second_vehicle_spawn_location)

        self.other_actors.append(first_vehicle)
        self.other_actors.append(second_vehicle)

By adding 1 to z-value as above, we can get the vehicle spawned at the required location.

closed 2019-04-03 by fabianoboril