carla-simulator/scenario_runner #106: Wrong behavior in object_crash_vehicle scenario

germanros1987, 2019-03-22:
We have detected an error in object_crash_vehicle

The reference to derive the location of the adversaries is obtained via self.ego_vehicle. 

```
lane_width = self._reference_waypoint.lane_width
location, _ = get_location_in_distance(self.ego_vehicle, _start_distance)
waypoint = self._wmap.get_waypoint(location)
```

However, this is incorrect for all those cases where the ego_vehicle is still far away from the scenario. The correct object would be config.ego_vehicle, which despite its horribly misleading name refers to the trigger position of the scenario.

I would advise to:

 * Change the parser to assign the right name to this field, i.e., scenario_trigger_location
* Fix this behavior

fabianoboril, 2019-03-22:
The name is not horribly wrong. It is designed to reflect the ego vehicle spawn location. We started to abuse the parser for things it was never designed to. Maybe we should more think about that, rather then just changing names...

fabianoboril, 2019-04-03:
Closed, as PR is merged.

PR #108 (2019-03-23): Feat trigger location scenario

#### Description

This PR updates the reference location of scenarios, that now is called trigger_point. It also fixes #106 

Fixes #106

#### Where has this been tested?

  * **Platform(s):** Ubuntu 16.04
  * **Python version(s):** 3.5.5
  * **Unreal Engine version(s):** NA
  * **CARLA version:** overnight 2019-03-19

#### Possible Drawbacks

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/108)
closed 2019-04-03 by fabianoboril