carla-simulator/scenario_runner #116: Throttle value does not drop down to zero after exceeding the target velocity in KeepVelocity Class

Dola47, 2019-03-26:
Dear All,

I just noticed after using KeepVelocity class for controlling both the Ego and the Othercar, that the throttle value does not drop down to zero after exceeding the set target_velocity.

Can any one inform me how to ensure that the throttle value drop down to zero?

I attached my edited code. I disable the scenario seq because I am not interested in the seq right now. I just want to be sure that the throttle value drop to zero.

Thanks.

[no_signal_junction_crossing.py.zip](https://github.com/carla-simulator/scenario_runner/files/3008309/no_signal_junction_crossing.py.zip)

mohit-awana, 2019-04-01:
Hi @Dola47, sorry for the late reply. In KeepVelocity behavior , the controlled traffic participant will accelerate as fast as possible until reaching a given _target_velocity_, which is then maintained for
as long as this behavior is active. To stop this behavior you need a termination behavior in parallel to this behavior.

Dola47, 2019-04-01:
@mohit-awana  Can you please provide me an example of how can this be implemented as a python code?

Dola47, 2019-04-09:
[no_signal_junction_crossing.py.zip](https://github.com/carla-simulator/scenario_runner/files/3058945/no_signal_junction_crossing.py.zip)

Hello All,

What I understand that KeepVelocity should make the car accelerate until reaching the target velocity then continue with this velocity. However in the attached code I defined the upper limit of the Target Velocities to 40, and the Cars still accelerating above that velocity.

Any Ideas why that is occurring, after the target velocity is reached the throttle value should be zero and the cars stop accelerating.

Thanks.

closed 2019-05-06 by Dola47