carla-simulator/scenario_runner #120: Resolve python requirements for scenario_runner

togaen, 2019-03-27:
The current docs state that scenario runner is designed to run on either python 2.7 or 3.5+, but this doesn't seem to be totally true. For example, the challenge evaluator will not run with python 2.7 (see #70). It looks like the code and the docs are inconsistent, so one or both need to be updated. I am happy to contribute either way, I would just need to know what the intended requirements are.

fabianoboril, 2019-03-27:
Hi,
we are currently fixing all these issues in preparation of the CARLA challenge. Please have a look on the development branch if it is urgent. Otherwise, I would recommend to wait for our next release, which arrive soon :)

togaen, 2019-03-27:
Great, thanks, does that mean the next release will support python 2.7? I am using ROS to drive my agent and 2.7 support would make that slightly easier :-).

fabianoboril, 2019-03-27:
It is our target to do so :)

lrmicmc, 2019-04-05:
Hi @togaen and @fabianoboril ,
we faced the same problem trying to use ROS and challenge_evaluator.py (master branch) with python 2.7 and we solved this problem by following the steps below:
+ challenge_evaluator.py
- replace: import importlib to import imp
- in method `__init__` replace lines 62-64 with: 
        `self.module_agent = imp.load_source(module_name, args.agent)`
- in method `_get_latlon_ref` replace line 297 with: 
        `if georef is not None:`

fabianoboril, 2019-06-17:
@lrmicmc @togaen : There is a new release 0.9.5 and also the master branch was updated. Everything should now be compatible with Python 2.7.

closed 2019-06-17 by fabianoboril