carla-simulator/scenario_runner #162: Track4 CARLA challeng

carlosalexander7, 2019-04-07:
I tried to run Track4SampleAgent but  when the pygame windows appears, the program is frozen. It seems that the run step function of the agent is never called. I used this command to run the agent:

```
python3 ${ROOT_SCENARIO_RUNNER}/srunner/challenge/challenge_evaluator_routes.py \
--scenarios=${ROOT_SCENARIO_RUNNER}/srunner/challenge/all_towns_traffic_scenarios3_4.json \
--routes=${ROOT_SCENARIO_RUNNER}/srunner/challenge/routes_training.xml \
--debug=0 \
--agent=${ROOT_SCENARIO_RUNNER}/srunner/challenge/autoagents/Track4SampleAgent.py
```

Has anyone successfully run the Track4SampleAgent ? 
Thanks in advance for any help :D

lijecaru, 2019-08-08:
Hi @carlosalexander7  any news regarding the Track4SampleAgent? I'm getting the same result when trying to run it, everything freezes and goes dark.

Yougmark, 2019-11-26:
I had a similar issue and worked around it by disabling synchronous mode.  It never runs into `run_step` indeed because it waits on the sensors or actors to be ready in a while loop forever.  I think this is fixed in the latest release.  Let me know if the problem still persists.  If not, please kindly close the issue.

mayankbhaskar007, 2019-12-30:
I am facing the same issue with the Track4SampleAgent, as the client freezes, in the latest release. How did you disable the synchronous mode?

Yougmark, 2019-12-31:
> I am facing the same issue with the Track4SampleAgent, as the client freezes, in the latest release. How did you disable the synchronous mode? @Yougmark

The synchronous mode is disabled in the latest master.  Which version are you using?

glopezdiest, 2021-11-23:
Closing the issue as this is no longer the case for current Scenario Runner.

closed 2021-11-23 by glopezdiest