carla-simulator/scenario_runner #172: error in scenario loading

Johnnyswong, 2019-04-09:
Hi, @nsubiron 
I have tried run the scenario_runner.py script, the command is:
`python scenario_runner.py --scenario FollowLeadingVehicle`.
The output is 
`Configuration for scenario FollowLeadingVehicle cannot be found!`

After jumped into your code, I found that the function named "find_scenario_config" in file "./srunner/tools/config_parser.py" may have some errors. It parse the xml file to some attributes and compare with the input scenario's name. But the attributes are named like "FollowLeadingVehicle_5", "FollowLeadingVehicle_6", etc.. No such attribute named "FollowLeadingVehicle".

Should I input with other arguments to specify the scenario?

Johnnyswong, 2019-04-09:
I switch the branch carla_challenge into master branch, the problem has gone.
It seems that the xml files is in a wrong form on the carla_challenge branch.

Johnnyswong, 2019-04-09:
But I always meets an timeout error. The log is:
`Preparing scenario: FollowLeadingVehicle
ScenarioManager: Running scenario FollowVehicle
timeout
Failure!
No more scenarios .... Exitin`

Johnnyswong, 2019-04-09:
I see. After the log of **ScenarioManager: Running scenario FollowVehicle**, must I add a ego car to following the car. 

Thanks a lot to create the scenario module, it may helps a lot.

closed 2019-04-09 by fabianoboril

ravishk1, 2019-04-17:
Hello @Johnnyswong  I am facing the same problem of:-

Preparing scenario: FollowLeadingVehicle
ScenarioManager: Running scenario FollowVehicle
Failure!
No more scenarios .... Exitin

I am not able to resolve this problem from your answer, Will you please elaborate a little more?
Thanks!!

Johnnyswong, 2019-04-17:
Hi @ravishk1 After the log:
Preparing scenario: FollowLeadingVehicle
ScenarioManager: Running scenario FollowVehicle

You must run an agent to connect the scenario sever in 10 seconds, otherwise it will cause a timeout. It seems that the default timeout setting is 10 seconds.

Try `python manual_control.py` after the log printed.

havefun28, 2019-07-02:
> After I checkout the carla_challenge branch into the master branch, the problem has gone.
> It seems that the xml file is in a wrong form on the carla_challenge branch.

Hi @ Thanks for your post and I met the same issue. However, switching to `master` branch does not solve my problem. Do u have any other tips? I am using CARLA0.9.5, Python3.5 on Ubuntu 18.04

havefun28, 2019-07-02:
> Hi! Thanks for your post and I met the same issue. However, switching to `master` branch does not solve my problem. Do u have any other tips? I am using CARLA0.9.5, Python3.5 on Ubuntu 18.04

Well, I figure it out by changing in `config_parser.py` around Line 220
```
# if set_attrib(scenario, 'name', None) == scenario_name:
if scenario_name in set_attrib(scenario, 'name', None):
```

Johnnyswong, 2019-07-03:
> > > After I checkout the carla_challenge branch into the master branch, the problem has gone.
> > > It seems that the xml file is in a wrong form on the carla_challenge branch.
> > 
> > 
> > Hi! Thanks for your post and I met the same issue. However, switching to `master` branch does not solve my problem. Do u have any other tips? I am using CARLA0.9.5, Python3.5 on Ubuntu 18.04
> 
> Well, I figure it out by changing in `config_parser.py` around Line 220
> 
> ```
> # if set_attrib(scenario, 'name', None) == scenario_name:
> if scenario_name in set_attrib(scenario, 'name', None):
> ```
> 
> same for Line 186 in `config_parser.py`
> 
> ```
> # if new_config.name == scenario_name:
> if scenario_name in new_config.name:
> ```
I think they have pushed the wrong code to the carla_challenge branch and the master branch....