carla-simulator/scenario_runner #238: Failed to localize before lane change

deepakreddy63, 2019-04-25:
Hi,

I wrote a script where other_actor merges with ego vehicle lane after moving ahead. It shows a warning 

> failed to Localize!: 54 -3
 Could you explain why this happens before lane change over?

Also, a general question. Why do we always get this "resetting ego-vehicle" message on the screen?

Thanks,
Deepak

Derekabc, 2019-05-11:
Hi, any update to this question?  Where can i find the lane merge algorithm? Thanks

deepakreddy63, 2019-08-08:
Hi,
@Derekabc .. I developed it using BasicAgent with setdestination.
This failed to localize issue happens when there is some issue with _localize function in global_route_planner.py script.

I'm closing this issue.Thanks.

closed 2021-11-23 by glopezdiest