carla-simulator/scenario_runner #239: Is there any lane-changing scenario in the scenario runner?

Johnnyswong, 2019-04-29:
Hi, the scenario runner for carla is very useful to benchmark a new algorithm. Many thanks to you guys. @nsubiron 
I was wondering, is there any lane-changing scenario in the scenario runner? I can't find it in the carla challenge branch.

Johnnyswong, 2019-04-29:
Also, are the "VehicleTurnLeftAtJunction1-VehicleTurnLeftAtJunction6" scenarios not implemented yet? I run the "python scenario_runner.py --scenario VehicleTurnLeftAtJunction_1" command, but get the following error:
`Scenario 'VehicleTurnLeftAtJunction' not supported ... Exiting`

lijecaru, 2019-05-10:
> Also, are the "VehicleTurnLeftAtJunction1" to "VehicleTurnLeftAtJunction6" scenarios not implemented yet?
> I ran the "python scenario_runner.py --scenario VehicleTurnLeftAtJunction_1" command, but got the following error:
> `Scenario 'VehicleTurnLeftAtJunction' not supported ... Exiting`

Hi Johnnyswong, did you find an answer for your question? I have been wondering the same for a while and didn't get a reply in a question that I posted.

Derekabc, 2019-05-15:
@Johnnyswong , @lijecaru same question for lane changing? Any update?

Johnnyswong, 2019-05-15:
No solution offered to solve this problem by the authors. But you can use
OtherLeadingVehicle1 to test the lane-changing case.

lijecaru, 2019-05-15:
> […](#)

Oh how do you use OtherLeadingVehicle1? In which branch is it? Thanks!

Johnnyswong, 2019-05-15:
In the carla challenge branch. Just run the scenario runner with arg
--scenario

Derekabc, 2019-05-22:
hi @lijecaru  any update on lane changing? need more detailed explanation. appreciate  any insights

fabianoboril, 2020-03-02:
We have a lane changing scenario, but in OpenSCENARIO description. Please see:
https://github.com/carla-simulator/scenario_runner/blob/master/srunner/examples/LaneChangeSimple.xosc

closed 2020-03-02 by fabianoboril