carla-simulator/scenario_runner #287: scenario_runner.py: IndexError: blueprint 'sensor.other.lane_detector' not found

ADulian, 2019-06-05:
Hello,

Just trying to run the latest master branch of scenario_runner and I get the following error message: 
"""
...
IndexError: blueprint 'sensor.other.lane_detector' not found
blueprint 'sensor.other.lane_detector' not found
No more scenarios .... Exiting
"""

Currently using carla 0.9.5. Any suggestion how to fix it?

Cheers

fabianoboril, 2019-06-07:
With CARLA 0.9.5 the name of this sensor has changed. You may want to switch to the development branch for now, or change the name to lane_invasion instead of lane_detector.

fabianoboril, 2019-06-17:
There is now a new release for scenario_runner to support 0.9.5. Also  the master branch was updated. This should fix this issue.

closed 2019-06-17 by fabianoboril