carla-simulator/scenario_runner #291: Error while Running all scenarios of one scenario class

deepakreddy63, 2019-06-10:
Hi,

I'm facing problem when running all scenarios 

python scenario_runner.py --scenario group:FollowLeadingVehicle

I get the following error 

deepak@advcs-Precision-Tower-5810:~/CARLA/scenario_runner$ python scenario_runner.py --scenario group:FollowLeadingVehicle
Preparing scenario: FollowLeadingVehicle_1
ScenarioManager: Running scenario FollowVehicle
Dist int success:  Actor(id=5357, type=vehicle.nissan.patrol)
Resetting ego-vehicle!
stopped: Actor(id=5357, type=vehicle.nissan.patrol)
Success!
Resetting ego-vehicle!
Trying to remove a non-existing actor id 5357
Preparing scenario: FollowLeadingVehicleWithObstacle_1
The scenario cannot be loaded
trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.
Preparing scenario: FollowLeadingVehicle_2
The scenario cannot be loaded
trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.
Preparing scenario: FollowLeadingVehicleWithObstacle_2
The scenario cannot be loaded
trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.
Preparing scenario: FollowLeadingVehicle_3
The scenario cannot be loaded
trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.
Preparing scenario: FollowLeadingVehicleWithObstacle_3
The scenario cannot be loaded
trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.
Preparing scenario: FollowLeadingVehicle_4
The scenario cannot be loaded
trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.
Preparing scenario: FollowLeadingVehicleWithObstacle_4
The scenario cannot be loaded
trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.
Preparing scenario: FollowLeadingVehicle_5
The scenario cannot be loaded
trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.

I'm not sure as to what might be the problem
I could locate to the location of the problem in scenario_runner.py file

self.prepare_ego_vehicle(config)

This line was causing issue.

Did anyone have similar problem. How to solve this? Thanks

deepakreddy63, 2019-06-10:
I feel that for each scenario configuration, the CarlaActorpool is being updated with the new world but the ego vehicle created for first scenario remains in the first scenario world which creates the below error

trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.

I tried to setup the ego vehicle for each scenario configuration and it works for same scenario class scenarios. But then same error come up in manual_control.py  program and need to restart the program for each continuing scenario instance.

fabianoboril, 2019-06-17:
Hi,

this is currently not possible, as the world is re-loaded after every scenario. A fix is in the pipeline. Stay tuned...

Best regards,
Fabian

deepakreddy63, 2019-06-17:
Good to know.
Waiting for the fix.
Thanks

fabianoboril, 2019-06-18:
Hi,
with PR #301 I added a commandline option --reloadWorld. Only when you use this flag the CARLA world is reloaded. In all other cases, the currently active world is used. This should resolve the described issue.

Please keep in mind: Scenarios that are not within the currently active world will be skipped.

rudrapoudel, 2019-06-18:
Hi @fabianoboril,

I have tried with PR #301 on 0.9.5 with added option --reloadWorld but got the following error,

`python scenario_runner.py --reloadWorld --scenario FollowLeadingVehicle_1`

Preparing scenario: FollowLeadingVehicle_1
ScenarioManager: Running scenario FollowVehicle
Resetting ego-vehicle!
Failure!
Resetting ego-vehicle!
ERROR: failed to destroy actor 527 : unable to destroy actor: not found 
No more scenarios .... Exiting

fabianoboril, 2019-06-18:
@rudrapoudel : It seems you run into a simple timeout of the scenario. Open manual_control.py and then run the scenario. You will see the cars being spawned and then you can drive around. Keep in mind that the timeout is based on passed time in CARLA, so if CARLA runs faster than realtime, the timeout may occur very soon!

rudrapoudel, 2019-06-19:
Hi @fabianoboril: I am using Carla server from here

>  http://carla-assets-internal.s3.amazonaws.com/Releases/Linux/Dev/CARLA_Latest.tar.gz

 and ran with following command` ./CarlaUE4.sh /Game/Carla/Maps/Town01 -benchmark -fps=20 -windowed`

Yes, I can run  manual_control.py successfully. Now, I have got following questions,

1. You said 'Open manual_control.py and then run the scenario'. How to do that I can't see any option for that. Or I got it wrong. When I tried to run both simultaneously manual_control.py crashed and same error with scenario_runner.py 
2. Is -fps=20 is the correct way to run CARLA  in realtime? Otherwise how? Where to modify?
3. Which version of server I should use- 0.9.5 or CARLA_Latest from above link? Which is the newest?
4. Also, if I crash/hit something with manual_control.py and do Ctrl+C or after with couple of calls. Sever takes forever to load the *.xodr maps and client just wait frequently with 'Scenario not yet ready message? This one also looks weird to me because if I restart the server than again it starts fine.

Thank you so much!

rudrapoudel, 2019-06-21:
I am really stopped here, any pointer on how to run a scenario would be great help.

deepakreddy63, 2019-06-21:
HI @rudrapoudel ,
1. You have a manual_control.py file in same scenario runner folder. Try running that file to get ego vehicle access of scenario runner scenario
4. If you crash/hit, ideally it should not affect and ctrl+c should kill the program right.

Hope this helps.

deepakreddy63, 2019-06-24:
Hi @fabianoboril 

Thanks for the update.

I had in mind to run different NHTSA scenarios in one xml file using group option.
I tried to use  --waitForEgo: Wait and connect option to do this and found that the ego vehicle faces problem to spawn first time as this argument is set and is stuck in infinite while loop in prepare_ego_vehicles function

So I modified as below
[image]

fabianoboril, 2019-06-24:
Hi,
--waitForEgo is intended to be used, if you spawn the vehicle outside of the scenario_runner (i.e. it exists already). Do you need to use this flag?

deepakreddy63, 2019-06-24:
Hi @fabianoboril ,

I Understand know. 

It seems in my case where I had to run different NHTSA scenarios defined in one config xml file using group option, I somehow ended up using your code/idea  to achieve this. 

Is it possible to add this kind of option/feature (run different scenarios in series one xml file)  in future release as standard like "group:" option.

Thanks

closed 2019-06-25 by fabianoboril

havefun28, 2019-07-02:
> Hi,
> with PR #301 I added a commandline option --reloadWorld. Only when you use this flag the CARLA world is reloaded. In all other cases, the currently active world is used. This should resolve the [...]

Hi, I follow your suggestion and add `--reloadWorld` flag as `python scenario_runner.py --reloadWorld --scenario group:ControlLoss`  but get the error
```
usage: scenario_runner.py [-h] [--host HOST] [--port PORT] [--debug]
                          [--output] [--file] [--junit] [--scenario SCENARIO]
                          [--randomize] [--repetitions REPETITIONS] [--list]
                          [--list_class] [-v]
scenario_runner.py: error: unrecognized arguments: --reloadWorld
```
Did you encounter similar problem?

havefun28, 2019-07-11:
> Hi @fabianoboril ,
> 
> I Understand now.
> 
> It seems in my case where I had to run different NHTSA scenarios defined in one config xml file using group option, I somehow ended up using this flag idea to achieve this.
> 
> Is it possible to add this kind of option/feature (run different scenarios in series in one xml file) in future release as standard like "group:" option?
> 
> Thanks

Can I ask whether you have successfully run a group of scenarios? If so, could you please share your way? I still get the error `RuntimeError: trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.` when I run `manual_control.py` with `python scenario_runner.py --scenario group:ControlLoss --reloadWorld` Thanks!

fabianoboril, 2019-07-12:
Please ensure that you use the correct CARLA version. Please have a look at the README.md:
https://github.com/carla-simulator/scenario_runner/blob/master/README.md#getting-the-scenariorunner

> It is important to also consider the release version that has to match the CARLA version.
Version 0.9.6: Compatible with CARLA 0.9.6
Version 0.9.5 and Version 0.9.5.1: Compatible with CARLA 0.9.5
Version 0.9.2: Compatible with CARLA 0.9.2

deepakreddy63, 2019-07-12:
Hi @havefun28 ,

I could run multiple scenarios in the same town and not spawning new ego_actor for each scenario with waitforego flag.

The above conditions ensure that we can use the manual_control.py file without the error.
Exact implementation is not with me right now.

cjerome0321, 2019-07-18:
Hi, @havefun28 
I met the same problem when I add --reloadWorld. I found a solution, after client.load_world(), do the following: 
```
    self.world = self.client.get_world()
    print(self.world.get_map())
    self.world = self.client.get_world()
```
I guess adding print ensures the client complete loading the map, after that the world could be accessed properly. I am not sure if this is the reason, but the error is gone.

havefun28, 2019-07-21:
>     self.world = self.client.get_world()
>     print(self.world.get_map())
>     self.world = self.client.get_world()

Hi, @cjerome0321 
Thanks a lot for your message and suggestions. However, I still get the same error after following your advice. May I check which file and/or function did you add the above chunk of code?
I tried to add it in `load_world()` method of `ScenarioRunner(object)` class. 
Carla version: 0.9.5

cjerome0321, 2019-07-22:
Hi, @havefun28 
I am using tag v0.9.5.1, in my case I added the code after client.load_world() [here](https://github.com/carla-simulator/scenario_runner/blob/v0.9.5.1/scenario_runner.py#L243).

havefun28, 2019-07-22:
> Hi, @havefun28
> I am using tag v0.9.5.1, in my case I added the code after client.load_world() [here](https://github.com/carla-simulator/scenario_runner/blob/v0.9.5.1/scenario_runner.py#L243).

Hi @cjerome0321 
Thanks a lot for your quick reply!

There is indeed update between 0.9.5 and 0.9.5.1 but unfortunately, I still get the error with `0.9.5.1` followings:
```
ERROR: trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.
Traceback (most recent call last):
  File "manual_control.py", line 622, in game_loop
    if not world.tick(clock):
  File "manual_control.py", line 158, in tick
    if len(self.world.get_actors().filter(self.vehicle_name)) < 1:
RuntimeError: trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "manual_control.py", line 681, in main
    game_loop(args)
  File "manual_control.py", line 630, in game_loop
    world.destroy()
  File "manual_control.py", line 177, in destroy
    actor.destroy()
RuntimeError: trying to access an expired episode; a new episode was started in the simulation but an object tried accessing the old one.
```
I may look into it but thanks a lot for your suggestions!:

cjerome0321, 2019-07-22:
> Hi @cjerome0321
> Thanks a lot for your quick reply!
>
> There is indeed update between 0.9.5 and 0.9.5.1 but unfortunately, I still get the error with `0.9.5.1` followings:
>
> ```
> ERROR: trying to access [...]

Hi @havefun28 ,
I added those codes just to make sure I could run scenario_runner.py successfully. I am not sure why you got the error in manual_control.py. I always run manual_control.py in another terminal after I run scenario_runner.py first and get:
```
Preparing scenario: ManeuverOppositeDirection_1
ScenarioManager: Running scenario ManeuverOppositeDirection
```

havefun28, 2019-07-22:
> Hi @havefun28 ,
> I added those codes just to make sure I could run scenario_runner.py successfully. I am not sure why you got the error in manual_control.py. I always run manual_control.py in another [...]
Hi @cjerome0321 
Thanks for your tips. I think I am doing the same... May I ask whether you built the CARLA from source?

cjerome0321, 2019-07-22:
> Hi @cjerome0321
> Thanks for your tips. I think I am doing the same... May I ask whether you built the CARLA from source?

Hi @havefun28 ,
Yes, but I built the CARLA on Windows 10...

PR #301 (2019-06-25): Support for multiple ego vehicles
#### Description
As one of CARLA's key features is the support for multiple ego vehicles, this PR brings in this feature also for the scenario execution. Multiple ego vehicles can be specified in the configuration file (XML) and identified via their unique role names. A simple example for a scenario using multiple ego vehicles is provided.

Other changes are:
* Added additional command line options for the scenario_runner.py
  - --waitForEgo: Wait and connect 
  - --outputDir: User defined output folder
  - --configFile: External XML file
  - --additionalScena [... cut]