carla-simulator/scenario_runner #299: Can't run Scenario_runner

dongyy1148, 2019-06-17:
OS: Ubuntu 16.04
Python: 3.5
I'm trying to run the : python scenario_runner.py
The problem is showed below. I use Carla 0.8.4 and newest py_trees packgae.
[image]

dongyy1148, 2019-06-17:
If I use python 2,7 to run the file, it will give me the result like
[image]
Could you help figure it out? Thanks!

fabianoboril, 2019-06-18:
Hi,

It seems you use an old CARLA version. For the scenario runner you need to use at least version 0.9.2, for the current version of the scenario runner even version 0.9.5 is required.

Regarding the Python 3.5 issue: It seems your py_tree installation has some issues. I recommend to install version 0.8.2 via pip, as described in the documentation.

dongzh1148, 2019-06-18:
Thanks a lot! I will try to implement with the latest version today!

fabianoboril, 2019-07-08:
I assume the problem was resolved. Please re-open the issue, if you still face issues.

closed 2019-07-08 by fabianoboril