carla-simulator/scenario_runner #302: Problem with repeated use of waypointFollower behavior

deepakreddy63, 2019-06-24:
Hi,

In "FollowLeadingVehicleWithObstacle" class, the WaypointFollower atomic behavior is used in two instances for same actor which causes the problem of vehicle looping back

> stop_near_intersection.add_child(WaypointFollower(self.other_actors[0], 35))
        stop_near_intersection.add_child(InTriggerDistanceToNextIntersection(self.other_actors[0], 20))

        driving_to_next_intersection.add_child(WaypointFollower(self.other_actors[0], self._first_actor_speed))
        driving_to_next_intersection.add_child(InTriggerDistanceToVehicle(self.other_actors[1],
                                                                          self.other_actors[0], 15))

deepakreddy63, 2019-06-24:
Hi,

I tried to correct this by making minor changes in WaypointFollower behavior.

[image]

I commented the setup function code and updated the waypoint list at the first update tick to get the latest waypoint list at current vehicle position.

krdindorkar19, 2019-07-05:
Hi @deepakreddy63 ,
I am also facing the same issue, you problem has been solved?

fabianoboril, 2019-07-05:
@deepakreddy63 @krdindorkar19 I just created a PR and a fix/waypointFollower branch. Please give it a try and let me know if it fixes the described issues.

krdindorkar19, 2019-07-06:
@fabianoboril Yes, it has been solved. Thank you !

fabianoboril, 2019-07-06:
@deepakreddy63 Your changes should not change the behavior,

closed 2019-07-06 by fabianoboril

deepakreddy63, 2019-07-08:
@fabianoboril  Yes, it works. Thanks !

PR #313 (2019-07-06): Fixed initialization of WaypointFollower behavior
Use initialise() method instead of setup().

Fixes #302

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/313)