carla-simulator/scenario_runner #305: Unable to spawn actors in carla: after pull

Dhairyashil-kpit, 2019-07-01:
HI,

BUG!!!!!!!!!!!!

carla version: 0.9.5

**1.  After taking pull of scenario runner branch, getting following error:** 

[image]

**2. From CarlaActorPool._world.try_spawn_actor() not able to spawn the actors into carla.**

3. Carla server is running as docker container with Town01 map

[image]

fabianoboril, 2019-07-01:
Can you connect to CARLA at all? Did CARLA throw a message why the vehicle could not be spawned? Which Python version do you use? Just writing "Bug!!!!!!!!" does not really help to debug the issue.

Dhairyashil-kpit, 2019-07-01:
> Can you connect to CARLA at all? Did CARLA throw a message why the vehicle could not be spawned? Which Python version do you use? Just writing "Bug!!!!!!!!" does not really help to debug the issue.

Hi @fabianoboril , I updated the post, please check.
I am not able to detect the issue why scenario runner is not able to spawn the actors using CarlaActorPool._world.try_spawn_actor().

fabianoboril, 2019-07-01:
Does CARLA inform you about a collision when trying to spawn an actor? If yes, try to use the flag --waitForEgo when starting

fabianoboril, 2019-07-01:
And could you please run it with --debug  to get some further output? Thank you.

fabianoboril, 2019-07-02:
Should be fixed with #307

fabianoboril, 2019-07-08:
I am closing this issue, as it seems resolved. Please re-open, if the problem is still there.

PR #307 (2019-07-02): Fixed spawn issue (collision with ground) for newer CARLA versions
Newer CARLA versions can cause a collision upon vehicle spawning when
the spawn point is on ground level.

Fixes #305 

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/307)
closed 2019-07-08 by fabianoboril