carla-simulator/scenario_runner #311: The scenario cannot be loaded?

JackRen88, 2019-07-03:
Hello.
when I run `python scenario_runner.py --scenario FollowLeadingVehicle_1`,I receive the following message:
`Preparing scenario: FollowLeadingVehicle_1
The scenario cannot be loaded
global name 'LocalPlanner' is not defined
No more scenarios .... Exiting
`
How I should solved it?

fabianoboril, 2019-07-05:
You need to setup the PYTHONPATH as described in the getting_started documentation.

JackRen88, 2019-07-08:
Hello, @fabianoboril .

Firstly,I setup the PYTHONPATH `(export PYTHONPATH=$PYTHONPATH:/home/rcj/carla_0.9.5/PythonAPI/carla/dist/carla-0.9.5-py2.7-linux-x86_64.egg:/home/rcj/carla_0.9.5/PythonAPI/carla/agents:/home/rcj/carla_0.9.5/PythonAPI)` in the .bashrc,when I run `python scenario_runner.py --scenario FollowLeadingVehicle_1`,I receive the following message:
`Traceback (most recent call last):
  File "scenario_runner.py", line 27, in <module>
    from srunner.scenarios.background_activity import *
  File "/home/rcj/scenario_runner-0.9.5/srunner/scenarios/background_activity.py", line 13, in <module>
    from srunner.scenariomanager.atomic_scenario_behavior import *
  File "/home/rcj/scenario_runner-0.9.5/srunner/scenariomanager/atomic_scenario_behavior.py", line 30, in <module>
    from srunner.tools.scenario_helper import detect_lane_obstacle, get_distance_along_route
  File "/home/rcj/scenario_runner-0.9.5/srunner/tools/scenario_helper.py", line 20, in <module>
    from agents.tools.misc import vector
ImportError: No module named agents.tools.misc
`
Therefore,I try to use the scripy(setup_environment.sh) that is in the scenario_runner to setup PYTHONPATH ,I get the message:
`Preparing scenario: FollowLeadingVehicle_1
The scenario cannot be loaded
global name 'LocalPlanner' is not defined
No more scenarios .... Exiting
`
How I should solved it? This problem has been bothering me for several days.

fabianoboril, 2019-07-08:
Please see the change in PR #318. This should resolve your issue. The PYTHONPATH was incomplete.

JackRen88, 2019-07-08:
Hello, @fabianoboril 
Thanks for your reply!
could you provide the complete path?

fabianoboril, 2019-07-08:
Which problem? The localplanner or the tools.misc is not found?

JackRen88, 2019-07-08:
The problem is that when I  include CARLA Python API to the Python path:
`export PYTHONPATH=$PYTHONPATH:/home/rcj/carla_0.9.5/PythonAPI/carla/dist/carla-0.9.5-py2.7-linux-x86_64.egg:/home/rcj/carla_0.9.5/PythonAPI/carla/agents:/home/rcj/carla_0.9.5/PythonAPI/carla`
I get the message:
`Preparing scenario: FollowLeadingVehicle_1 The scenario cannot be loaded global name 'LocalPlanner' is not defined No more scenarios .... Exiting`

fabianoboril, 2019-07-08:
Try the following:
`export CARLA_ROOT=/home/rcj/carla_0.9.5
 export PYTHONPATH=${CARLA_ROOT}/PythonAPI/carla:${CARLA_ROOT}/PythonAPI/carla/dist/carla-0.9.5-py2.7-linux-x86_64.egg:${CARLA_ROOT}/PythonAPI/carla/agents`
Then run Python (Python 2.7 in this case) and run:
`from agents.navigation.local_planner import LocalPlanner`

Does this work?

JackRen88, 2019-07-09:
@fabianoboril 。
I don't understand your instructions.
I look at the scenario_runner.py file,find that exception is 
`try:
                    CarlaActorPool.set_world(self.world)
                    self.prepare_ego_vehicle(config)
                    scenario = scenario_class(self.world,
                                              self.ego_vehicle,
                                              config,
                                              args.randomize,
                                              args.debug)
                except Exception as exception:
                    print("The scenario cannot be loaded")
                    if args.debug:
                        traceback.print_exc()
                    print(exception)
                    self.cleanup()
                    continue`
Please help me find where the problem is.

fabianoboril, 2019-07-09:
Well for some reason, your python environment is not complete. Therefore, please run the steps I posted above (just one after the other...), such that we can try to understand where the problem is exactly.

JackRen88, 2019-07-10:
@fabianoboril.
According to your  instruction.
I run the following step:

fabianoboril, 2019-07-10:
Using the same export commands, can you try to run the scenario?

JackRen88, 2019-07-11:
Hello,@fabianoboril .
I can't run the scenario.

`rcj@rcj-OMEN-by-HP-Laptop-15-ce0xx:~/scenario_runner-0.9.5$ python scenario_runner.py --scenario FollowLeadingVehicle_1
Preparing scenario: FollowLeadingVehicle_1
The scenario cannot be loaded
global name 'LocalPlanner' is not defined
No more scenarios .... Exiting`

fabianoboril, 2019-07-11:
Can you please run with "--debug". Then we will see the full exception. Something seems wrong with the import for your setup.

fabianoboril, 2019-11-26:
I assume the issue was resolved as there was no further input. Please re-open if it still exists.

closed 2019-11-26 by fabianoboril

laks1806, 2022-07-11:
@fabianoboril, I tried changing the path and everything seems correct but this error keeps coming : 
```
python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld
Configuration for scenario FollowLeadingVehicle_1 cannot be found!
No more scenarios .... Exiting
```
my python path is: 
`/opt/carla-simulator/PythonAPI/carla/dist/carla-0.9.11-py2.7-linux-x86_64.egg:/opt/carla-simulator/PythonAPI/carla/agents:/opt/carla-simulator/PythonAPI/carla`

PR #318 (2019-07-10): Fixed PythonPath in documentation
The getting_started documentation contained a non-completed PYTHONPATH.

Resolves #311 
Fixes #319 

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/318)