carla-simulator/scenario_runner #312: Scenario Runner : New scenario addition

krdindorkar19, 2019-07-04:
Hi,

I am trying to create a new scenario, kindly refer the attached file : 
[lane_cut_in.txt](https://github.com/carla-simulator/scenario_runner/files/3359177/lane_cut_in.txt)

I was just experimenting around it, I found one thing, my stage_1, stage_2, and stage_3 are same where I am using WaypointFollower and DriveDistance behavior.

I have added this stages in a sequence tree and tried to run it.

My Observation : 
My vehicle is working fine till stage_2, however for stage_3 it is reversing and coming back to it's previous location and starting stage_3 behavior from that position.

Kindly help me to know why this is happening.

fabianoboril, 2019-07-05:
Is this related to #302?

krdindorkar19, 2019-07-06:
@fabianoboril, yes, I will close it. Thanks for you help

closed 2019-07-06 by krdindorkar19