carla-simulator/scenario_runner #323: Not expected result when running challenge_evaluation

deepakreddy63, 2019-07-10:
Hi,
I started the Carla server both with no editor. Later, I also tried with the unreal editor.
Then, I tried running as per the documentation,
python srunner/challenge/challenge_evaluator_routes.py --scenarios=${ROOT_SCENARIO_RUNNER}/srunner/challenge/all_towns_traffic_scenarios1_3_4.json --agent=${ROOT_SCENARIO_RUNNER}/srunner/challenge/autoagents/DummyAgent.py

and saw no output from sensors in the terminal.
I checked the results.json file and below is the image of sample part of file.
I understood that the error is due to Carla simulator not responding. But then it responds when I run normal scenario_runner.py scenario to verify the problem.

[image]

I'm not sure what might be the problem. Is it possible to try with a new file with a small number of scenarios?

Thanks,
Deepak

fabianoboril, 2019-07-11:
Which version of CARLA and the scenario runner do you use?

deepakreddy63, 2019-07-11:
I use 0.9.5 Carla version and 0.9.5 scenario runner (latest with git pull).

fabianoboril, 2019-07-11:
Attention: If you are on master branch and do a git pull, you have a version that does no longer work with CARLA 0.9.5. This is already a preparation for the upcoming CARLA 0.9.6. Please checkout the version v0.9.5.1. This should work with CARLA 0.9.5. I just updated the documentation to clarify this point.

deepakreddy63, 2019-07-11:
Hi,

Thanks for the reply.
I tried switching to carla_challenge branch and seemed to work fine in one system.

when I tried in another system it gives time out error.
[image]

Any idea as to why this is happening?

Thanks,
Deepak

fabianoboril, 2019-07-12:
Hi,
you don't need to switch to the old challenge branch. Just go back to the tagged version v0.9.5.1. This should work with CARLA 0.9.5. 

Sometime you can get timeout failures. This depends a bit on the system performance on how responsive CARLA is on this system.

deepakreddy63, 2019-07-16:
HI @fabianoboril ,

I switched to tagged version v0.9.5.1 and tried the script.
I get the following errors for dummy agent where it says wheel doesn't have attribute steer_angle.

[image]

The pygame window appears and dissapear  for HumanAgent
[image]

It shows same error for carla_challenge branch for DummyAgent, but HumanAgent works though.

Thanks

fabianoboril, 2019-07-17:
Then you use a CARLA version that is newer than 0.9.5. At some point steer_angle became max_steer_angle...

deepakreddy63, 2019-08-08:
Hi,
I tries with v0.9.6 of both scenario_runner and carla to run challenge_evaluator_routes.py script. 

Just a small correction. 
There is no input for "rolename" parameter in request_new_actor function inside "prepare_ego_vehicle" function of challenge_evaluator_routes.py file and this causes the ego_vehicle to not move when NPCAgent is used as it cannot find rolename attribute for ego actor in NPCAgent script.

fabianoboril, 2019-08-08:
Could you please test PR #345? I just fixed this issue...

deepakreddy63, 2019-08-08:
The fix works and ego vehicle is now moving.

But I noticed a few more issues when ego vehicle moves.
1) Scenario 3 doesn't spawn human actor
2) Ego vehicle abruptly stops at [34.99] of route1 in town 02

Not sure what is happening

lijecaru, 2019-08-09:
@deepakreddy63  how did you try the fix? I'm also testing NPC agent but it doesn't move and I don't find where to modify it. Thank you.

fabianoboril, 2019-08-09:
@deepakreddy63: Please see PR #346. This should fix the first problem.

deepakreddy63, 2019-08-09:
@fabianoboril 
Thansk for the fix. The second problem also seems to be having the same correction as first one where the InTriggerDistanceToLocationAlongRoute is not triggered as it is taking self.other_actors[0].get_location() as input instead of self._other_actor_transform.location.

fabianoboril, 2019-08-09:
Oh yes. Fix is on its way.

fabianoboril, 2019-08-09:
PR #347 addresses TS04. Please let me know, if it helps.

deepakreddy63, 2019-08-19:
Hi @fabianoboril ,

A small correction.
In PR 347, it is changed as self._other_actor_transform instead of self._other_actor_transform.location.

fabianoboril, 2019-08-20:
Fix will be released soon.

PR #356 (2019-08-20): Enable routes as scenarios
#### Description
- Enhanced stability of scenario execution (Fixes #323 #355)
- Added ability to execute routes as scenarios via ScenarioRunner
- Split scenario configs and parser into different modules for better comprehension
- Updated documentation

Checklist:
  - [x] Your branch is up-to-date with the `master` branch and tested with latest changes
  - [x] Extended the README / documentation, if necessary
  - [x] Code compiles correctly and runs
  - [x] Code is formatted and checked with Utilities/code_check_and_formatting.sh
  - [x] Changelog is updated

#### Whe [... cut]
closed 2019-08-20 by fabianoboril