carla-simulator/scenario_runner #324: Overtaking Scene using OpenDrive

jS5t3r, 2019-07-16:
The scenario manager can support OpenDrive, but not all features: https://github.com/carla-simulator/scenario_runner/blob/master/Docs/openscenario_support.md   Can I write an overtake  scenario by using positions so that the car follows them?  Lateral action is not now supported.

fabianoboril, 2019-07-17:
Hi,
you could define waypoints to achieve an overtake scenario. This is already supported. Another option would be to extend the current implementation to feature lateral actions ;-)

YDSh510, 2019-07-23:
Hi,

Just wanted to understand more on OpenSceanrio as not much is given in below link http://www.openscenario.org/.
What are we trying to achieve using openscenario. does that mean from now we don't have to have the regular scenario file . bit confused. please help.

jS5t3r, 2019-07-25:
You can define a scenario in a xml.file. Carla can supports that partly.

fabianoboril, 2019-07-29:
> Hi,
> 
> Just wanted to understand more on OpenSceanrio as not much is given in below link http://www.openscenario.org/.
> What are we trying to achieve using openscenario. does that mean from now we don't have to have the regular scenario file . bit confused. please help.

Instead of describing the scenario programmatically in Python (as it was done at the beginning) you can now define a scenario in an XML file. In the long run, this will be much more comfortable and flexible!

jS5t3r, 2019-08-01:
In the zip, is the overtaking scenario with lateral support:
http://www.openscenario.org/docs/OpenSCENARIO_v0.9_examples.zip

Path: OpenSCENARIO_v0.9/examples/Overtaker/Overtaker.xosc

There is also a video: http://www.vires.com/OpenSCENARIO/Videos/Example5.mpeg

kino3, 2019-08-19:
@fabianoboril Is there any list of milestones for OpenSCENARIO tags? It is important to decide whether I will wait for your contribution or not (try to implement by myself).

fabianoboril, 2019-08-19:
We are working on such a public schedule. Which feature are you looking for?

kino3, 2019-08-19:
@fabianoboril I'd like to use "Lateral" tag to express a cut-in scenario. Can we use waypoints or something instead? #334 shows that "Lateral" tag will be available soon...

jS5t3r, 2019-09-13:
Ok. I will wait for the lateral support. I am going to use waypoints.

fabianoboril, 2019-11-26:
Some examples are now available. See for instance: https://github.com/carla-simulator/scenario_runner/blob/master/srunner/examples/LaneChangeSimple.xosc

closed 2019-11-26 by fabianoboril

anna-srl, 2020-04-18:
Hi @fabianoboril ,

What schema did you use in order to run the above xosc with scenario runner?
I tried to run it using the ones already available in scenario runner (srunner>openscenario) but there are several points where they differ, it seems it was not built based on that..
I'm using Carla 0.9.5 and scenario runner 0.9.5.1, however, I checked the schema available in other versions and they don't seem to differ a lot..
Many thanks in advance!