carla-simulator/scenario_runner #325: Error while following Getting Started doc

wangwei14, 2019-07-19:
Hi, I just followed all the steps in the Getting Started documentation, but it prompts errors again and again after running scenario_runner.py. The message was "Timestamp does not have any field named frame" and it was in srunner/scenariomanager/timer.py. I already installed Python packages, set the Python path, and everything. I looked into the code and found out there is timestamp.frame in manual_control.py, which worked fine after the server was up. Did I miss any steps?

fabianoboril, 2019-07-29:
This is a mismatch in versions. It seems you use an old CARLA version with the most recent scenario_runner. Please have a look into our README on compatible versions.

wangwei14, 2019-07-30:
I tried both 0.9.4 and 0.9.5 of CARLA but both of them did not work.

fabianoboril, 2019-07-30:
It is important to also consider the release version that has to match the CARLA version.

    Version 0.9.6: Compatible with CARLA 0.9.6
    Version 0.9.5 and Version 0.9.5.1: Compatible with CARLA 0.9.5
    Version 0.9.2: Compatible with CARLA 0.9.2

wangwei14, 2019-08-04:
Thanks a lot. I will try that.

fabianoboril, 2019-08-21:
I assume the issue was resolved. Please re-open the issue if it still exists.

closed 2019-08-21 by fabianoboril