carla-simulator/scenario_runner #328: Synchronous Mode

Dola47, 2019-07-25:
Dear All,

I would like to ask, if it is possible to run the scenarios in the synchronous mode.

If it is possible, which modifications should be done to the manual_control.py to do so.

If there is an example already done it will be so helpful.

Thanks.

fabianoboril, 2019-07-29:
The synchronous mode for the scenario execution is not that simple, as you require 2 clients (one client is the scenario engine, and the second client controls the ego vehicle). Currently, having 2 synchronous clients is not directly supported by CARLA. Hence, you would need to implement a basic synchronization between these two clients, and then you could enable the synchronous mode inside the scenario_runner (see the challenge_evaluator_routes as example).

Dola47, 2019-08-07:
@fabianoboril  it is not really very clear for me how it is implemented in challenge evaluator, can you please provide me with a clear example how synchronous mode can be implemented in the scenarios case.

yyab, 2020-09-14:
> Dear All,
> 
> I would like to ask, if it is possible to run the scenarios in the synchronous mode.
> 
> If it is possible, which modifications should be done to the manual_control.py to do so.
> 
> If there is an example already done it will be so helpful.
> 
> Thanks.

I also have the same question. Thanks!

glopezdiest, 2021-11-23:
In order to run scenarios in synchronous mode, you can now use the `--sync` flag at scenario_runner.py. This will make the simulation run synchronously but remember that only one client can do the world.tick and the others (such as the manual control) will have to wait for it (so they'll run asynchronously)

closed 2021-11-23 by glopezdiest

VidaAPI, 2023-06-23:
> o run scenarios in synchronous mode, you can now use the `--sync` flag at scenario_runner.py. This will make the simulation run synchronously but remember that only one client can do the world.tick and the others (such as the manual control) will have to wait f

@fabianoboril  @glopezdiest I need to get and plot the velocity of the adversary vehicle in follow leading vehice. Using TM requires --sync. since i need to record them too i also use --record . Meanwhile for getting and plotting the velocity i am trying to make a new metrics, which requires the TM stuff, which is only for sync mode. Now i am very much confused how to do it. I couldnt find any helpful tips unfortunattly.