carla-simulator/scenario_runner #334: Can we define present Traffic light and Intersection scenarios in OpenSCENARIO?

deepakreddy63, 2019-07-31:
Hi @fabianoboril ,

I was trying to implement the traffic light scenarios in current Traffic scenarios in openSCENARIO format and was confused as to how to include traffic lights or intersection details in the openScenario file. I don't see any support for these in openSCENARIO schema.
Any help is appreciated.

Also, can we use Waypoint follower for longitudinal action in openSCENARIO by adding terminal conditions similar to KeepVelocity atomic behavior. This will make vehicle follow the path as keepVelocity is causing vehicle to move out of lane sometimes.

Thanks
P.S: I could add Lateral action from my side extending the current OpenSCENARIO features

fabianoboril, 2019-08-01:
Hi @deepakreddy63,

great to hear that you can support the development. This is always very welcome. The standard process is that you open a pull request for your addition, and then we review this PR. So, if you have something already at hand for lateral actions, we are more than happy to integrate it.

Regarding the traffic lights. OpenSCENARIO has conditions that make use of States (ByState), and one of them uses Signals.
`
                        <xsd:element name="Signal">
                            <xsd:complexType>
                                <xsd:attribute name="name"  type="xsd:string" use="required"/>
                                <xsd:attribute name="state" type="xsd:string" use="required"/>
                            </xsd:complexType>
                        </xsd:element>
`

The waypoint follower is already used for Routing actions :)

Bests

deepakreddy63, 2019-08-01:
Hi @fabianoboril,

I'm glad to contribute. 
How do I create a pull request as I don't have write access to scenario_runner repository. Can you give any suggestions as to how to proceed with this.

Thanks

fabianoboril, 2019-08-01:
Hi,
you can fork the repo into your own github space, update the code there and then create a PR here :)

jS5t3r, 2019-11-24:
Can the traffic lights edited via Infrastructure or Roadnetwork? When I start the scenario, I want to have the traffic lights in an exact state.

fabianoboril, 2019-11-26:
Hi,

this will work inside a maneuver section:

                         <Event name="TrafficLight" priority="overwrite">
                            <Action name="TrafficLight">
                                <Global>
                                    <Infrastructure>
                                        <Signal>
                                            <SetState name="158" state="green" />
                                        </Signal>
                                    </Infrastructure>
                                </Global>
                            </Action>
                            <StartConditions>
                                <ConditionGroup>
                                    <Condition name="StartCondition" delay="0" edge="rising">
                                    <ByEntity>
                                        <TriggeringEntities rule="any">
                                            <Entity name="hero" />
                                        </TriggeringEntities>
                                        <EntityCondition>
                                            <TraveledDistance value="1.0" />
                                        </EntityCondition>
                                    </ByEntity>
                                    </Condition>
                                </ConditionGroup>
                            </StartConditions>
                        </Event>

jS5t3r, 2019-12-27:
Hi, 
Thanks.  Almost working. Every time, when I start: ` <SetState name="158" state="green" />, where 158 is the actor id. Every time, when I start OpenSCENARIO (python scenario_runner.py --openscenario srunner/examples/traffic.xosc --reloadWorld), the traffic lights have different IDs. Aren't there any global IDs?

```
    def show_closest_trafficlights(self):
        def print_trafficlightlist(traffic_lights):
            print("==========================================")
            print("=== Prints a list of closest traffic lights: ")
            for tf in traffic_lights:
                print(tf[0], tf[1].id, tf[1].type_id)
            print("==========================================")

        self.actors_list = self.world.get_actors().filter('traffic.traffic_light*')
        closest_traffic_lights = []
        for actor in self.actors_list:
            d = euclidean_dist(self.vehicle, actor)
            closest_traffic_lights.append((d, actor))

        closest_traffic_lights.sort(key=itemgetter(0))
        print_trafficlightlist(closest_traffic_lights)
```

fabianoboril, 2019-12-28:
Hi,
the IDs change if CARLA is not properly restarted. Maybe this should be addressed at the CARLA main repo.
We used the other work around a while ago as well. However, I would prefer to ensure that the traffic light IDs do not change (at the moment I recommend to restart the CARLA server).

fabianoboril, 2021-09-24:
IDs are meanwhile supported.

closed 2021-09-24 by fabianoboril

ssap09, 2024-07-16:
Hi,
I am trying to use carla and import an osc file to simulate the scenario. I am not able to find any osc file which has traffic signal information, Can anyone please share an example osc file which has traffic signals information which can be imported to carla?
Any help is appreciated. 
Thanks in advance.