carla-simulator/scenario_runner #342: Failure when using scenario_runner.py or challenge_evaluator_routes.py

erwef, 2019-08-06:
After attempting to run the scenario runner challenge evaluation, following the instructions in https://github.com/carla-simulator/scenario_runner/blob/v0.9.5.1/Docs/challenge_evaluation.md

I used the following command:

`python srunner/challenge/challenge_evaluator_routes.py --scenarios=%ROOT_SCENARIO_RUNNER%/srunner/challenge/all_towns_traffic_scenarios1_3_4.json --agent=%ROOT_SCENARIO_RUNNER%/srunner/challenge/autoagents/DummyAgent.py`

However, this gives some weird behaviour which doesn't look right to me. Each scenario only lasts 1-4 seconds, and throws multiple errors (even skipping some scenarios).

The console output:
```
WARNING: attempting to destroy an actor that is already dead: Actor 5176 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 5402 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 5625 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 5853 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 6068 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 6289 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 6511 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 6855 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 7232 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 7603 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 7950 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 8315 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 8689 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 9048 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 9414 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 9779 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 10122 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 10400 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 10659 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 10935 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 11199 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 11467 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 11721 (vehicle.lincoln.mkz2017)
Base transform is blocking objects  Transform(Location(x=8.06553, y=-99.3817, z=0.250015), Rotation(pitch=0, yaw=90.0863, roll=0))
Base transform is blocking objects  Transform(Location(x=7.86552, y=-99.382, z=0.250015), Rotation(pitch=0, yaw=90.0863, roll=0))
Base transform is blocking objects  Transform(Location(x=7.66552, y=-99.3823, z=0.250015), Rotation(pitch=0, yaw=90.0863, roll=0))
Base transform is blocking objects  Transform(Location(x=7.46553, y=-99.3826, z=0.250015), Rotation(pitch=0, yaw=90.0863, roll=0))
Base transform is blocking objects  Transform(Location(x=7.26553, y=-99.3829, z=0.250015), Rotation(pitch=0, yaw=90.0863, roll=0))
Base transform is blocking objects  Transform(Location(x=7.06553, y=-99.3832, z=0.250015), Rotation(pitch=0, yaw=90.0863, roll=0))
Skipping scenario 'Scenario4' due to setup error: Error: Unable to spawn vehicle vehicle.diamondback.century at Transform(Location(x=7.06553, y=-99.3832, z=0.250015), Rotation(p
itch=0, yaw=90.0863, roll=0))
WARNING: attempting to destroy an actor that is already dead: Actor 11985 (vehicle.lincoln.mkz2017)
Base transform is blocking objects  Transform(Location(x=-212.749, y=-8.43287, z=0.30545), Rotation(pitch=0, yaw=449.861, roll=0))
Base transform is blocking objects  Transform(Location(x=-212.949, y=-8.43238, z=0.30545), Rotation(pitch=0, yaw=449.861, roll=0))
Base transform is blocking objects  Transform(Location(x=-213.149, y=-8.43189, z=0.30545), Rotation(pitch=0, yaw=449.861, roll=0))
Base transform is blocking objects  Transform(Location(x=-213.349, y=-8.43141, z=0.30545), Rotation(pitch=0, yaw=449.861, roll=0))
Base transform is blocking objects  Transform(Location(x=-213.549, y=-8.43092, z=0.30545), Rotation(pitch=0, yaw=449.861, roll=0))
Base transform is blocking objects  Transform(Location(x=-213.749, y=-8.43043, z=0.30545), Rotation(pitch=0, yaw=449.861, roll=0))
Skipping scenario 'Scenario4' due to setup error: Error: Unable to spawn vehicle vehicle.diamondback.century at Transform(Location(x=-213.749, y=-8.43043, z=0.30545), Rotation(p
itch=0, yaw=449.861, roll=0))
WARNING: attempting to destroy an actor that is already dead: Actor 12265 (vehicle.lincoln.mkz2017)
WARNING: attempting to destroy an actor that is already dead: Actor 12559 (vehicle.lincoln.mkz2017)
```

Also, trying to replay the episode from the logs using this command:
`python start_replaying.py -f log_dev_track3_route_0024.log -c 7375`
Crashes the Carla server with the error "fatal error", I've tried this using multiple log files.

I also tried running a single scenario using the commands:
`(venv) C:\CARLA_0.9.5\PythonAPI\scenario_runner-0.9.5>python scenario_runner.py --scenario FollowLeadingVehicle_1 --output --file`
and
`C:\CARLA_0.9.5\PythonAPI\scenario_runner-0.9.5>manual_control.py --autopilot`

This gives the following console output:
`Preparing scenario: FollowLeadingVehicle_1
ScenarioManager: Running scenario FollowVehicle
Resetting ego-vehicle!

Scenario: FollowVehicle --- Result: FAILURE
Start time: 2019-08-06 18:26:39
End time: 2019-08-06 18:27:12
Duration: System Time 32.09s --- Game Time 57.90s
Ego vehicle:  Actor(id=528, type=vehicle.lincoln.mkz2017)
Other actors: Actor(id=529, type=vehicle.nissan.patrol);

                Actor             |            Criterion           |   Result    | Actual Value | Expected Value
-----------------------------------------------------------------------------------------------------------------
         lincoln.mkz2017 (id=528) |         CheckCollisions (Req.) |        INIT |         0.00 |         0.00
                                  |                       Duration |     SUCCESS |        57.90 |        60.00

Failure!
Resetting ego-vehicle!
No more scenarios .... Exiting
`

I'm not sure why it is attempting to reset the ego-vehicle or states a failure occurred as there was no collision in the simulation.

I'm using the following setup:
 - OS: Windows 10
 - CARLA Version 0.9.5
-  scenario runner 0.9.5
 - Python 3.7
 - Unreal Engine 4.22.3

fabianoboril, 2019-08-21:
Hi,

let's go through your comments one by one:
- "Resetting ego-vehicle!": This is a CARLA output when using the local_planner (see https://github.com/carla-simulator/carla/blob/528a59319b8715dfbf9447a10a0a1d5d28e871fa/PythonAPI/carla/agents/navigation/local_planner.py). This has nothing to do with the ego vehicle, as defined by the scenario. This is only an output, the planning module, which we use in some behaviors (e.g. to make an adversary vehicle follow some waypoints). Maybe the verbosity of the local_planner should be configurable, to avoid this irritating output.
- Result is FAILURE for FollowLeadingVehicle scenario: This is due to the fact that the CheckCollision test is in Init-State not Success-State. I assume you use an older version of the ScenarioRunner, as this was fixed already a few weeks ago.
- Challenge execution: The output you see is not about that scenarios are lasting only for a few seconds. The output is generated when creating the scenarios and something goes wrong during the creation. For example, at certain locations collisions during spawning of actors can occur. In this case the scenario is removed, which you can see in your log.
```
Skipping scenario 'Scenario4' due to setup error: Error: Unable to spawn vehicle vehicle.diamondback.century at Transform(Location(x=7.06553, y=-99.3832, z=0.250015), Rotation(p
itch=0, yaw=90.0863, roll=0))
```

siddancha, 2019-10-02:
I have a similar issue with CheckCollision test being INIT. Here's my console output --

```console
(base) siddharthancha@yertle:~/repos/scenario_runner$ python scenario_runner.py --scenario FollowLeadingVehicle_1 --output
Preparing scenario: FollowLeadingVehicle_1
ScenarioManager: Running scenario FollowVehicle
Resetting ego-vehicle!

Scenario: FollowVehicle --- Result: FAILURE
Start time: 2019-10-01 23:25:59
End time: 2019-10-01 23:26:30
Duration: System Time 31.03s --- Game Time 30.54s
Ego vehicle:  Actor(id=1333, type=vehicle.lincoln.mkz2017)
Other actors: Actor(id=1334, type=vehicle.nissan.patrol); 

                Actor             |            Criterion           |   Result    | Actual Value | Expected Value 
-----------------------------------------------------------------------------------------------------------------
         lincoln.mkz2017 (id=1333) |         CheckCollisions (Req.) |        INIT |         0.00 |         0.00 
                                  |                       Duration |     SUCCESS |        30.54 |        60.00 

Failure!
Resetting ego-vehicle!
No more scenarios .... Exiting

```
I am using the latest version of the `scenario_runner` repository.

fabianoboril, 2019-10-04:
Are you controlling the ego vehicle, for example with the manual_control.py? If not, this is the expected outcome.

siddancha, 2019-10-04:
> Are you controlling the ego vehicle, for example with the manual_control.py? If not, this is the expected outcome.

Thanks for replying! Yes, I am controlling the vehicle using `manual_control.py`. I succeed in other criteria, but CheckCollisions is always init.

fabianoboril, 2020-03-02:
This was fixed. Please update to a newer version for example 0.9.7

closed 2020-03-02 by fabianoboril