carla-simulator/scenario_runner #349: DynamicObjectCrossing scenario uses incorrect filter for Walker blueprints

napratin, 2019-08-13:
**Description**
[`DynamicObjectCrossing._spawn_adversary()`](https://github.com/carla-simulator/scenario_runner/blob/9d04ca800929adcf496ab853a72a44955c8cb31a/srunner/scenarios/object_crash_vehicle.py#L202) uses the following line to randomly select and spawn a "walker" actor:
```python
walker = CarlaActorPool.request_new_actor('walker*', transform)
```

This filter (`'walker*'`) inadvertently allows the (newly added?) `'controller.ai.walker'` actor type to be selected (randomly), which doesn't expose the same API as a vehicle or walker, and leads to a crash when the `HandBrakeVehicle` behavior is executed in the scenario's behavior tree.

**To Reproduce**
1. Setup to run Carla server version 0.9.6 with matching client installed/available on Python path.
2. Clone the `scenario_runner` repo and checkout tag `v0.9.6` (and make sure it is available on Python path).
3. Run the `DynamicObjectCrossing` scenario multiple times.
4. Eventually, you should hit an error like the following:
```
...
File "scenario_runner/srunner/scenariomanager/atomic_scenario_behavior.py", line 1180, in update
    self._vehicle.apply_control(self._control)
AttributeError: 'WalkerAIController' object has no attribute 'apply_control'

During handling of the above exception, another exception occurred:
...
File "scenario_runner/srunner/scenariomanager/atomic_scenario_behavior.py", line 563, in terminate
    self._control.throttle = 0.0
AttributeError: 'NoneType' object has no attribute 'throttle'
```

**NOTE**: Since this error is triggered at random, you may have to run the scenario a bunch of times to actually trigger the error. An alternate way is to simply query the blueprint library with the same filter (`'walker*'`), and you will notice (`'controller.ai.walker'`) included as one of the blueprints.

**Expected behavior**
Scenario should execute without errors every time.

**Platform**
 - OS: Ubuntu 16.04
 - CARLA version: 0.9.6
 - Python version: 3.5.2
 - Version: 0.9.6

**Solution**
A simple solution that effectively solves this is to change the filter from `'walker*'` to `'walker.*'`:
```python
walker = CarlaActorPool.request_new_actor('walker.*', transform)
```

fabianoboril, 2019-08-14:
Thank you for this amazingly described report. I'll fix this with the next PR.

closed 2019-08-14 by fpasch

napratin, 2019-08-14:
@fabianoboril Thanks for the prompt fix! Any chance this can be "back-ported" to `v0.9.6`? (i.e. the tag moved up to the latest commit, or a new `v0.9.6.1` release tag added)

Not a big deal - happy to continue using `master` for now; just want to make sure it remains compatible with client/server `0.9.6`.

fabianoboril, 2019-08-15:
We will add a new tag, if something becomes incompatible with 0.9.6. As long as things remain compatible with 0.9.6 we just continue on master.

napratin, 2019-08-23:
I realized a related issue is that parts of `scenario_runner` repo are themselves changing, and some of that is leading to incompatibilities. Case in point is a recent commit that moved/renamed some files (https://github.com/carla-simulator/scenario_runner/commit/3c1bf4a0d747c11123c507c88c01701d29d03a33) including `config_parser` -> `scenario_config_parser`, which I was trying to use to parse scenario XML in my own way.

Now, I need to use the walker fix (which is on master), but I don't want any of these other changes; and I anticipate such situations to arise in the future as well. Any advice would be appreciated, as I do want to stay updated with changes. Maybe a separate versioning/sub-versioning scheme for the `scenario_runner` repo so that it is not only tied to Carla versions? (If not, I'll just keep my local copy on a desired commit.)

fabianoboril, 2019-08-26:
Hi,
we started having internal discussions as well on this topic. For now however, the master branch shows always the most recent version designed to work with the latest CARLA (release). If you have some relevant changes, you might consider bringing it back to github via a pull request. Otherwise, you need to cherry-pick your changes (or the fixes you would like to have from github).

Reasons for the changes are to have a better architecture and organization for the next CARLA challenge iteration, and new OpenSCENARIO features :)

PR #351 (2019-08-14): Fixed walker selection in DynamicObjectCrossing
Instead of requesting an actor of type walker* the type walker.* has to
be used to avoid using the 'controller.ai.walker' by accident.

Checklist:
  - [x] Your branch is up-to-date with the `master` branch and tested with latest changes
  - [x] Extended the README / documentation, if necessary
  - [x] Code compiles correctly and runs
  - [x] Code is formatted and checked with Utilities/code_check_and_formatting.sh
  - [x] Changelog is updated

Fixes #349 

#### Where has this been tested?
  * **Platform(s):** Ubuntu 16.04
  * **Python version(s): [... cut]