carla-simulator/scenario_runner #350: ScenarioRunner on Windows 10 can work?

kino3, 2019-08-14:
Is there anyone who can work ScenarioRunner with Windows 10 environment?
[Getting Started](https://github.com/carla-simulator/scenario_runner/blob/master/Docs/getting_started.md#installing-prerequisites) just says that It designed to be used in Ubuntu, not says **cannot work** with Windows.

I tried to use ScenarioRunner in my Windows 10 environment for recent days. I can install any related packages and can run `python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld`, but this script ends with `Failure!`.

If someone uses this plugin on Windows 10, I will continue my struggles.
But some developer says officially "This engine cannot work with Windows 10 for now", I will set up Ubuntu environment asap :-)

fabianoboril, 2019-08-14:
There is a similar issue #342 
I never tested on Windows, so I can't tell you exactly if it works.

Did you run the manual_control.py after starting the scenario? You need to connect an "agent" to the ego vehicle to make the scenario do something meaningful.

kino3, 2019-08-16:
Thank you. After that, I can connect the ego vehicle by manual_control.py and can execute some basic scenarios, though pythonAPI puts out the following unknown errors repeatedly.
`
OverflowError: can't convert negative value to unsigned int
Traceback (most recent call last):
  File "C:\Users\Shuji\Documents\carla\scenario_runner-0.9.5.1\srunner\scenariomanager\scenario_manager.py", line 205, in _tick_scenario
    self.scenario_tree.tick_once()
  File "C:\Users\Shuji\AppData\Local\Programs\Python\Python37\lib\site-packages\py_trees\behaviour.py", line 158, in tick_once
    for unused in self.tick():
  File "C:\Users\Shuji\AppData\Local\Programs\Python\Python37\lib\site-packages\py_trees\composites.py", line 578, in tick
    for node in child.tick():
  File "C:\Users\Shuji\AppData\Local\Programs\Python\Python37\lib\site-packages\py_trees\composites.py", line 578, in tick
    for node in child.tick():
  File "C:\Users\Shuji\AppData\Local\Programs\Python\Python37\lib\site-packages\py_trees\behaviour.py", line 249, in tick
    new_status = self.update()
  File "C:\Users\Shuji\Documents\carla\scenario_runner-0.9.5.1\srunner\scenariomanager\atomic_scenario_criteria.py", line 477, in update
    closest_waypoint_right = self._map.get_waypoint(sample_point_right, lane_type=carla.LaneType.Any)
`

closed 2019-08-16 by kino3

TKeutgens, 2020-09-15:
@kino3 Hi, im running into the same negative value error. Did you resolve this issue?

kino3, 2020-09-17:
No, recently I only use Ubuntu instead.