carla-simulator/scenario_runner #357: Better ROS support

cmpute, 2019-08-20:
Currently the ros_agent.py are basically a simple version of carla ros-bridge. Could you please change it to one that fully ultilize carla ros-bridge? For example, the transformations between frames are not published yet in the script.

cmpute, 2019-08-21:
And also, the sign of y location is a big problem for us

fabianoboril, 2019-08-21:
Hi,

the current form of the ros_agent was intended to be used for the CARLA AD Challenge. As a consequence, the amount of information provided by this agent is very limited to ensure that participants cannot access ground truth data. Hence, the ros_agent offers only a significantly reduced feature set compared to the ros-bridge.

The question is, if you require something that is comparable to the ros-bridge, why not using the ros-bridge directly? You could basically have an agent that starts the ros-bridge (via start.sh as in the ros_agent) and then only does some minimal control work.

cmpute, 2019-08-21:
Thanks for your reply, I think I would try to implement an agent using that existing code

closed 2019-08-21 by cmpute