carla-simulator/scenario_runner #360: fps issue

deepakreddy63, 2019-08-22:
**Description of question or idea**
I upgraded scenario runner to 0.9.6  from 0.9.5
I see that scenarios run faster in 0.9.6 version when compared to 0.9.5 and look like speedup version of that when running in 0.9.5. 
Is this an fps issue? 
I saw that frame_rate is added and set to 20 in scenario runner and also updated with each reload world. 
How to set fps for simulator using config,py, if it gets updated with each reloadWorld when we run a new scenario.
Also, the HUD shows the server and client fps as varying and near to 60. Should anyone of them show 20 as we are setting the value to 20?

**Expected behavior**
Normal execution of scenario with no speedup.

**Desktop (please complete the following information):**
 - OS:Ubuntu 16.04
 - CARLA Version CARLA 0.9.6
 - Python version 3.6
 - Version  0.9.6

fabianoboril, 2019-08-26:
Hi,

0.9.6 is faster than 0.9.5 (I think this was somehow related to the switch to Unreal 4.22). We have a synchronous mode on the roadmap for scenario runner, which would help to address your issue.

matthewcoyle-cpc, 2019-11-27:
Probably worth adding an fps command line argument

deepakreddy63, 2020-02-03:
Hi @fabianoboril ,

Still looking for this feature in scenario_runner as it will allow us to run the scenario at the desired speed. Sometimes scenario runs way faster and would like to play at a slower speed.
Thanks !!

> Hi,
> 
> 0.9.6 is faster than 0.9.5 (I think this was somehow related to the switch to Unreal 4.22). We have a synchronous mode on the roadmap for scenario runner, which would help to address your issue.

fabianoboril, 2020-02-04:
The synchronous mode works with agents (like in the CARLA challenge). If this is not what you need, you can take a look into the ScenarioManager code, where the main loop is called:
https://github.com/carla-simulator/scenario_runner/blob/master/srunner/scenariomanager/scenario_manager.py#L221
Here you can skip frames, or wait for frames, if you require this.

deepakreddy63, 2020-02-05:
Thanks for the reply.

I just have a question, Is the server FPS shown in manual_control.py HUD display related to the fps setting we set from config.py file. I don't see any change even after setting the fps to different values.