carla-simulator/scenario_runner #365: Improve logical compliance with OpenSCENARIO intent

matthewcoyle-cpc, 2019-08-27:
**Description**
The current behaviour tree interpretation of OpenSCENARIO executes acts, maneuvers and events sequentially as well as ignoring ConditionGroups

**Expected behavior**
Acts, Maneuvers and Events should occur in parallel
ConditionGroups should be used to group conditions to allow for logical ORs

matthewcoyle-cpc, 2019-08-27:
@fabianoboril I've made the changes on my forked branch https://github.com/matthewcoyle-cpc/scenario_runner/tree/openscenario_logic
Unfortunately I'm still in the position where I can't test the changes at all

fabianoboril, 2019-08-27:
@matthewcoyle-cpc: Can you rebase on the current master branch and open a PR? I will test your changes in parallel, and then we could start bringing in the first set of new OpenSCENARIO features 👍

PR #366 (2019-09-16): Openscenario logic
#### Description

All sequences of actions now occur in parallel
ConditionGroups allow for multiple conditions to be evaluated in parallel for logical ORs

Fixes #365 Improve logical compliance with OpenSCENARIO intent

#### Where has this been tested?

This has not yet been tested

#### Possible Drawbacks

This could change the behaviour of existing scenarios including the order of events and triggering of conditions

---
This change is [[image]](https://reviewable.io/reviews/carla-simulator/scenario_runner/366)
closed 2019-09-16 by fabianoboril