carla-simulator/scenario_runner #367: Scenario Runner on windows

Faabdoly, 2019-08-28:
Hi,
In getting start doc  you said "The current version is designed to be used with Ubuntu".
Do you intend next version compatible with windows?
or Can I use current version in windows with change in command that you mentioned in getting start?
thanks a lot for your attention.

fabianoboril, 2019-08-30:
Hi,
some folks are using ScenarioRunner also under Windows. So it seems to work, but we are currently not actively working on Windows support.

anna-srl, 2020-03-04:
Hi,

thank you for all your amazing work!

not sure if I should comment here or open a new issue. 

I was trying to use scenario runner under windows 10, however I couldn't figure out how to apply the steps, are you still not providing support? Specifically I am not able to figure out how to to include CARLA Python API to the Python path (I'm quite new to both carla and python).

In general my intention was to use carla in order to run specific scenarios (specifically overtaking scenarios, like the one mentioned in https://github.com/carla-simulator/scenario_runner/issues/324), do you have any suggestions on other ways to do that?

I am using the pre-packaged version of carla 9.5.

Thank you very much for your time.

Faabdoly, 2020-03-07:
ADD environment variable Like that:
Environment Variables->System Variables
add CARLA_ROOT  Variable and Variable value of it like this:
C:\...\Carla_0_9_5

LikhitaNukala, 2020-05-14:
Please share setup instructions for scenario runner on Windows also. The Getting started documentation  seems to be specific to Linux. I have downloaded Carla 0.9.8 source code and scenario runner 0.9.8. I have built that but did not find any egg files. Please give more instructions pertaining to Windows.

Ramachetan, 2020-05-14:
I'm also trying scenario runner on windows but unable to start please give me steps in order to run scenario runner on windows

NVrlab, 2020-05-17:
Python API Build in Windows:
1.Delete Build folder in Project Path.
2.Check that CmakeList.txt  file exist in project path.( if  that file changed in build CarlaUE 
process copy and replace it from source)
3.update setuptools
4.run make PythonAPI or python setup.py install in Visual Studio x64 Native Tools Command Promps 
setup.py is in PythonAPI\carla  folder.
note: there is version.h that saved with ini format that must be changed  its format to .hpp.
it is in …\carla\LibCarla\source\carla.
this steps Done in carla 0.9.7

Faabdoly, 2020-05-17:
Python API Build in Windows:
1.Delete Build folder in Project Path.
2.Check that CmakeList.txt  file exist in project path.( if  that file changed in build CarlaUE 
process copy and replace it from source)
3.update setuptools
4.run make PythonAPI or python setup.py install in Visual Studio x64 Native Tools Command Promps 
setup.py is in PythonAPI\carla  folder.
note: there is version.h that saved with ini format that must be changed  its format to .hpp.
it is in …\carla\LibCarla\source\carla.
this steps Done in carla 0.9.7

LikhitaNukala, 2020-06-16:
Thanks for the reply. Also, can you please tell me how to set the Python path in Windows. In Linux, it is path of the egg file. What should it be for Windows?

Faabdoly, 2020-06-17:
> Thanks for the reply. Also, can you please tell me how to set the Python path in Windows. In Linux, it is path of the egg file. What should it be for Windows?

ADD environment variable Like that:
Environment Variables->System Variables
add PYTHONPATH Variable and Variable value of it like this:
...\Carla_0_9_5\PythonAPI\carla\dist\carla-0.9.5-py3.7-win-amd64.egg
and
...\Carla_0_9_5\PythonAPI\carla\agents
and
...\Carla_0_9_5\PythonAPI\carla

glopezdiest, 2020-06-26:
Hey @Faabdoly @LikhitaNukala @Ramachetan @anna-srl.

We just created a PR with some Docs explaining how to set up the environment variables. We also know that the manual control had an issue with the selected font, and that's fixed too.

Have any of you found any other problems regarding support for Windows (missing / incorrect Docs, files not working...)?

glopezdiest, 2020-06-29:
Closing this issue as we just merged a PR with this explanation at the Docs (PR #581)

closed 2020-06-29 by glopezdiest