carla-simulator/scenario_runner #376: FollowLeadingVehicleWithObstacle_9 in Town05 exhibits unexpected behavior

napratin, 2019-09-05:
**Describe the bug**
I observed some unexpected behavior when running scenario `FollowLeadingVehicleWithObstacle_9` (type="FollowLeadingVehicleWithObstacle", town="Town05"), defined in [FollowLeadingVehicle.xml](https://github.com/carla-simulator/scenario_runner/blob/master/srunner/examples/FollowLeadingVehicle.xml#L55), as depicted below:

[image]

The vehicle (actor) that our ego is supposed to follow suddenly performs a hard right turn and goes off-road, rendering the rest of the scenario useless. It's hard to reproduce this behavior as it only happens randomly. I have also observed similar incidents when running other scenarios (especially FollowLeadingVehicle ones). Not sure where the fault lies - whether in the scenarios, behavior trees, or controller that controls these actors, so any help would be appreciated.

**To Reproduce**
With a Carla server (0.9.6) running (with options: `-benchmark -fps=20`):
1. Run the following scenario (may need `--reloadWorld` if not in `Town05`):
```bash
python scenario_runner.py --scenario FollowLeadingVehicleWithObstacle_9
```
2. Run some controller on ego vehicle, e.g. `manual_control.py` or `no_rendering_mode.py`, to make it move (this should trigger the scenario to start). 
3. Follow along behind the actor and observe its behavior. 
4. Unexpected behavior is observed at random; you may have to repeat the above steps several times to trigger it.

**Expected behavior**
Actor should proceed forward as planned, and not go off the road. In this scenario, it is supposed to go forward, wait for the cyclist to cross, and then continue.

**Screenshots**
(GIF included above.)

**Desktop (please complete the following information):**
 - OS: Ubuntu 16.04
 - CARLA Version: 0.9.6
 - Python version: 3.5
 - Version: 0.9.6

fabianoboril, 2019-09-06:
Hi,

there is indeed a problem with the WayPointFollower (LocalPlanner in CARLA). It seems that this agent from time to time produces strange behavior. We are looking into this and hope to find a solution soon.

fabianoboril, 2019-11-11:
Hi,
we think we solved this issue with #PR395. Please test it again. I will close this issue for now. Please re-open it, if the problem still exists.

closed 2019-11-11 by fabianoboril