carla-simulator/scenario_runner #387: Definition of "Position" by roadId, s and laneId

asanebub, 2019-09-27:
It's very difficult to spawn the entities by global coordinates. In the most cases the entitites can't be spawned to the defined coordinates.

So it would be an easement to spawn the entities directly to the roadId and laneID.
Also it's very important for the Action "Lane change".

fabianoboril, 2019-09-27:
Agree. Do you plan to work on this topic, or is this just an identification of the gap?

asanebub, 2019-09-28:
I'm working on scenario generation for highly automated driving and it's very difficult for me to replicate my defined scenarios, because I can't spawn the objects to the desired location. 

It would be very helpful if you implement this.

fabianoboril, 2019-10-21:
Hi,
just a short update. The CARLA API is going to be extended with a function that will provide us a mapping from roadId, laneId, s to a CARLA waypoint. Once this method is there, we can start adding support for your request.

kino3, 2019-12-11:
I would be happy to tell you when that feature will be implemented. I didn't find any official schedule or milestone about Carla new API on GitHub.

fabianoboril, 2020-03-04:
Started working on this in branch feature/roadIdOSC. Still experimental, but might work with CARLA master.

fabianoboril, 2020-03-10:
Merged today with #483

closed 2020-03-10 by fabianoboril