carla-simulator/scenario_runner #399: In OpenSCENARIO defined command driving backward should be supported.

jS5t3r, 2019-11-04:
@fabianoboril Please, help me for that.

I want to define negative velocity in the OpenSCENARIO scenario as in line 124: https://gist.github.com/jS5t3r/c1e2543ede9275b0c735b5100d3ef7c0

But the car does not move backwards. Probably, it must be programmed that by a negative velocity 
the car sets its reverse gear. For example, by a negative velocity, a pedestrian can walk backwards.

matthewcoyle-cpc, 2019-11-05:
From our perspective it's sensible that negative speeds should mean 'reverse' as there's no restriction on that number stating that it must be positive and there's no other reasonable interpretation of negative speed values

jS5t3r, 2019-11-11:
Would you plan this feature in near future?

fabianoboril, 2019-11-11:
Hi,
we plan to change internally a couple of things to be able to use the new TrafficManager of CARLA, that was recently merged and will be announced with the next release. Afterwards, we can look in how to enable going backwards, as this will change some of the underlying behaviors quite significantly (e.g. lane following will become better, easier to use, etc.)

closed 2020-10-01 by fabianoboril

bapukokare, 2022-10-11:
@fabianoboril   can you please guide me to support -ve velocity, in my case vehicle is initiated at high speed with respect to the waypoint and time information. and parameter declaration is used to start with a defined velocity.  can we move the car strictly with waypoint and time information during deceleration and stop a vehicle at some interval

PR #646 (2020-10-01): Not allowing negative target speeds

#### Description

Current situation: Vehicle does not move when negative target speeds are used.

Throws an exception for vehicle controllers `npc_vehicle_control` and `pedestrian_control` when it encounters a negative target speed.
Using `simple_vehicle_control` or `vehicle_longitudinal_control` allows for negative speeds, and the vehicle will reverse as expected.

Fixes #399  

#### Where has this been tested?

  * **Platform(s):** Ubuntu 18.04
  * **Python version(s):** 2.7
  * **Unreal Engine version(s):** 4.24
  * **CARLA version:** 0.9.9.4

#### Possi [... cut]