carla-simulator/scenario_runner #400: Failed to run challenge_evaluator_routes.py with HumanAgent

linxigjs, 2019-11-06:
When i run challenge_evaluator_routes.py as Docs/challenge_evaluation.md says, 
`python srunner/challenge/challenge_evaluator_routes.py --scenarios=${ROOT_SCENARIO_RUNNER}/srunner/challenge/all_towns_traffic_scenarios1_3_4.json --agent=${ROOT_SCENARIO_RUNNER}/srunner/challenge/autoagents/HumanAgent.py
`
there is a black window named "Human Agent" appears and disappears after a while, then repeatedly appear and disappears. What's wrong with that? What should i do to solve it?
carla version: v0.9.5, scenerio runner version: v0.9.5.1
[image]

ZTige, 2019-11-08:
I have the same issue, have you figured it out? Thanks

Yougmark, 2019-11-24:
I am able to run HumanAgent with the master branch and my own small fix #408.
It's pretty cool.  Don't forget to turn on the debug option for viewing the waypoints visualization.

linxigjs, 2019-11-26:
> I am able to run HumanAgent with the master branch and my own small fix #408.
> It's pretty cool. Don't forget to turn on the debug option for viewing the waypoints visualization.

Thank you for your help, but my problem is still unresolved. And in my test, no exception threw, which is different from your #408

Yougmark, 2019-11-26:
@linxigjs Are you still using 0.9.5 scenario runner? I had issues with synchronous mode with older version (0.9.6 I think).  Probably the problem was there in 0.9.5 too.  You can try disabling synchronous mode in `challenge_evaluator_routes.py`.  I remember it was running forever in a while loop waiting for the sensors to be ready or actors to be created (not sure which one).

Anyway, try tracing down the `challenge_evaluator_routes.py` script and see where it gets stuck.

linxigjs, 2019-12-03:
Thanks for your answer @Yougmark , I will figure it out once I have the time...

glopezdiest, 2021-11-23:
Closing this as it is no longer relevant to current Scenario Runner. Feel free to reopen it if that isn't the case

closed 2021-11-23 by glopezdiest