carla-simulator/scenario_runner #419: What does "RuntimeError: frame == _episode->GetState()->GetTimestamp().frame" mean?

GilgameshD, 2020-01-07:
**Describe our question or idea**
When I run the challenge mode (routescenario), I sometimes get the error when call world.click(). The scenario running may not be influenced by it, but I am wondering why this happens.

**Expected behavior**
No error appears

**Screenshots**

Traceback (most recent call last):
    File "/home/wenhao/scenario_runner/srunner/scenariomanager/scenario_manager.py", line 272, in _tick_scenario
    CarlaDataProvider.get_world().tick()
RuntimeError: frame == _episode->GetState()->GetTimestamp().frame

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - CARLA Version CARLA 0.9.6
 - Python version 3.7
 - Version  0.9.6

fabianoboril, 2020-01-10:
Hi,
when CARLA cannot perform the next tick due to whatever reason, CARLA raises an exception:
https://github.com/carla-simulator/carla/blob/4738f734de84f9547a6c30e70c4b840ecc43181c/LibCarla/source/carla/client/detail/Simulator.cpp#L122

Is this a reproducible problem?

GilgameshD, 2020-01-11:
Hi,
thanks for the information! It is reproducible, but there is only about 20% probability that this exception happens. I am still collecting more information about it. I will close this issue now and reopen it when I can provide more details!

closed 2020-01-11 by GilgameshD