carla-simulator/scenario_runner #428: Hardcode trigger positions in scenario 10 (no_signal_junction_crossing)

GilgameshD, 2020-01-18:
**Describe the bug**
Four trigger positions in no_signal_junction_crossing.py are hardcoded, which cannot work with configuration files.

I don't know if there is any plan to implement this no_signal_junction_crossing scenario with position coordinates that come from configuration files. It will help a lot if you can implement it!

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - CARLA Version: CARLA 0.9.6
 - Python version: 3.7

fabianoboril, 2020-01-19:
Currently we moved our focus more towards the OpenSCENARIO-based scenarios. If you need a configurable version of this scenario soon, you may want to try it out yourself? If you struggle with getting it working, we can give you some assistance and support.

GilgameshD, 2020-01-19:
Thanks, I will try it out myself. And I am very looking forward to the OpenSCENARIO format scenario, I think it is more convenient to define scenarios in that way.

closed 2020-01-19 by GilgameshD