carla-simulator/scenario_runner #429: How to generate my own route and config files?

GilgameshD, 2020-01-20:
**Describe our question or idea**
I notice that the challenge route (dev and test) does not trigger many scenarios and some scenario locations are not very reasonable. Ex: in Town05, scenario3 only happens on the freeway, where the pedestrians are blocked by the guard bar.

So is there any tool to generate my own route and scenario config files in a convenient way? Thanks!

fabianoboril, 2020-01-21:
Hi,

in the challenge subfolder, you can find a *.json file containing the scenario locations in each town. You can edit this file to fit your requirements. For the locations, I recommend to drive along the route you want to evaluate and write down the x/y-coordinates of potential scenario locations.

GilgameshD, 2020-01-21:
Thanks a lot

closed 2020-01-21 by GilgameshD