carla-simulator/scenario_runner #430: Spawning bug of some vehicles when running route scenarios

glopezdiest, 2020-01-22:
**Describe the bug**

When running a route_scenario, part of the vehicels spawned in Town01 and Town06 instantly deviate from the route and crash. This si caused only in some specific lanes of the towns and the vehicles always aim for the same direction after being spawned.

**To Reproduce**

Steps to reproduce the behavior:
1. Run an scenario_route

**Expected behavior**

Normal spawn of the vehicles.

**Screenshots**

Link to a screen cast where the [bug](https://we.tl/t-WhTuZmomET) is shown

**Desktop (please complete the following information):**
 - OS: Ubuntu 0.9.7
 - CARLA Version 0.9.7
 - Python version 3.6
 - Version 0.9.6

fabianoboril, 2020-01-23:
Hi,
would it be possible to check this also with CARLA 0.9.6? I am seeing issues with the new TrafficManager in 0.9.7...

fabianoboril, 2020-01-23:
With 0.9.7 I can see this problem as well, with 0.9.6 it does not seem to be happening. However, the control of the background traffic is through the CARLA autopilot, which in 0.9.7 is the TrafficManager. I guess we have to look in there.

glopezdiest, 2020-01-23:
This doesn't happen when running a "normal" simulation without scenario_runner though, so that means it isn't relate to the TrafficaManager, right?

fabianoboril, 2020-01-23:
Could be something that the input to the TM is not as it is in the normal simulation.

fabianoboril, 2020-03-02:
This is an issue with the TM. Will be fixed with #469

closed 2020-04-09 by glopezdiest