carla-simulator/scenario_runner #443: Loop Behavior with OpenSCENARIO

heobu, 2020-02-05:
I would like to be able to create a loop behavior for vehicles using OpenSCENARIO. It seems like already there might be some potential solutions to this problem, but none of them seem to work so far.

In these scenarios, each vehicle has a dedicated Act describing their behavior inside one over-arching Story for all the vehicles. And inside each Act, a _Sequence>Maneuver>Event>Action>Private>Routing>FollowRoute>Route_ is described consisting of a bunch of Waypoints so that the vehicle can follow along a given route for this scenario.  Firstly, looking at [the complementary support documentation for OpenSCENARIO](https://github.com/carla-simulator/scenario_runner/blob/master/Docs/openscenario_support.md) and looking through the code, the expectation was that increasing the **numberOfExecutions** parameter for a _Sequence_ tag would ensure this whole Sequence would be repeated. But, be it a _Route_ or any other type of scenario definition, changing the numberOfExecutions parameter does not seem to have any effect whatsoever and the Sequence never gets repeated. Secondly, as an alternative to the first method, additional Events or appropriate conditions are used to create an Act in which one specific Event consisting of a Route is supposed to be triggered again and again at the end of each loop. Seemingly, even with more basic scenario definitions, an Event cannot be triggered more than once to provide this looping behavior anyway. As a result, that approach did not provide the desired result either.  

Environment:
 - OS: Ubuntu 16.04
 - CARLA Version: CARLA 0.9.7
 - Scenario Runner Version: master (commit id: ffd8f8c1d51344bf8037233006725eda654df6a4)

fabianoboril, 2020-02-05:
Hi,
thanks for this hint. I could reproduce the problem. There is in fact a bug. I started looking into a solution.

fabianoboril, 2020-02-05:
You can take a look into the following branch:
https://github.com/carla-simulator/scenario_runner/tree/fix/repetitions
For my quick tests it seems to fix the issues.

heobu, 2020-02-10:
Hi. This fix seems really helpful and indeed now you can see the vehicles attempting to repeat the same behavior. Except the fact that when it is used in a Sequence involving Routing>FollowRoute>Route. Because now the vehicle will complete the first loop, slow down and stop, make a small loop to go back a bit and start the second loop only then. As a result, the transition between these repetitions is awkward and has some extra undefined behavior.

heobu, 2020-02-10:
A small addition to my previous comment: This is at least the case with a Speed parameter set using ParameterDeclaration. Without a speed set, so far it seems to be working in repeating Sequences with smooth transitions in between.

heobu, 2020-02-10:
Another point: In a case where multiple cars are following their Routes under one over-arching Story, when the Ego car (hero) completes its loop, the other cars start behaving strangely and lose their way for a bit as the Ego vehicle is said to be reset and destroyed:
> Resetting ego-vehicle!
> Destroying ego-vehicle!

Visually, the Ego vehicle does not seem to be disappearing between these loops anyway, and in between these repetitions for the Ego vehicle, it seems like maybe the other cars are trying to find their way again to possibly continue following their original route. But somehow, the Ego vehicle completing 1 execution of a Sequence has some sort of side-effect on the other vehicles.

fabianoboril, 2020-02-10:
Hi,
the last of your posts refers to the CARLA LocalPlanner:
https://github.com/carla-simulator/carla/blob/94dab0c4d7e16d349a0c2ff8d205cf546ad32cca/PythonAPI/carla/agents/navigation/local_planner.py#L91
This creates this not-so-nice debug output. You can ignore it, or file an issue at CARLA.

We are currently performing the review for the PR, and plan to merge it. We will look into the problem with the speed parameter afterwards.

heobu, 2020-02-10:
Hi,

About the last point on the local planner, the issue is not that there is a debug output being printed by it, it’s that once a vehicle is finished with following its route, all the other vehicles seem to get confused about their own route. This seems more of an issue with waypoint following or scenario management actually.

fabianoboril, 2020-03-02:
If all are using the same local planner module, this is as I explained above. Do you see this also when using different routes for different vehicles?

commit 627ec147 (2020-02-06): Fixed repetitions in OpenSCENARIO (Fixes #443)

Repetitions in OpenSCENARIO were not properly working. Reworked the repetition
handling by three major changes:
- Not copy py_tree elements, this will cause problems due to references in Python
- Replaced get_py_tree_path with get_xml_path, as the first was not working properly
- Avoid using ":" in py_tree behavior names, as this may break py_trees

Change-Id: I7829106136cd6c20e6863d1f424fbc600030bf13
commit f97b10ba (2020-02-06): Fixed repetitions in OpenSCENARIO (Fixes #443)

Repetitions in OpenSCENARIO were not properly working. Reworked the repetition
handling by three major changes:
- Not copy py_tree elements, this will cause problems due to references in Python
- Replaced get_py_tree_path with get_xml_path, as the first was not working properly
- Avoid using ":" in py_tree behavior names, as this may break py_trees

Change-Id: I7829106136cd6c20e6863d1f424fbc600030bf13
commit 82aa03bd (2020-02-06): Fixed repetitions in OpenSCENARIO (Fixes #443)

Repetitions in OpenSCENARIO were not properly working. Reworked the repetition
handling by three major changes:
- Not copy py_tree elements, this will cause problems due to references in Python
- Replaced get_py_tree_path with get_xml_path, as the first was not working properly
- Avoid using ":" in py_tree behavior names, as this may break py_trees

Change-Id: I7829106136cd6c20e6863d1f424fbc600030bf13
commit 1aef2d8f (2020-02-10): Fixed repetitions in OpenSCENARIO (Fixes #443)

Repetitions in OpenSCENARIO were not properly working. Reworked the repetition
handling by three major changes:
- Not copy py_tree elements, this will cause problems due to references in Python
- Replaced get_py_tree_path with get_xml_path, as the first was not working properly
- Avoid using ":" in py_tree behavior names, as this may break py_trees

Change-Id: I7829106136cd6c20e6863d1f424fbc600030bf13