carla-simulator/scenario_runner #445: cannot run demo scenarios and error message is displayed

aqvia, 2020-02-05:
**Description**
I cannot run demo scenarios. Instead, a message is displayed on the console.
It looks like there is a mismatch in the argument types of the PythonAPI.

**To Reproduce**
Executed around 15:00 on 2020-02-04.
1. Download CARLA_0.9.7.3.tar.gz (compiled version) and extract it.
2. Clone "carla-simulator/scenario_runner".
3. Follow the instractions on 'Docs/getting_started.md'.
   1. Install prerequisites(For Python 3.x)
   2. Include CARLA Python API to the PYTHONPATH
4. Start the CARLA server: `./CarlaUE4.sh`
5. Open another tarminal and execute: `python scenario_runner.py --openscenario srunner/examples/FollowLeadingVehicle.xosc --reloadWorld`
6. See the following message:
```
...
Boost.Python.ArgumentError: Python argument types in
WeatherParameters.init(WeatherParameters)
did not match C++ signature:
init(_object, float cloudiness=0.0, float precipitation=0.0, float precipitation_deposits=0.0, float wind_intensity=0.0, float sun_azimuth_angle=0.0, float sun_altitude_angle=0.0, float fog_density=0.0, float fog_distance=0.0, float wetness=0.0)
init(_object)
...
```

**Desktop:**
 - OS: Ubuntu 18.04.1
 - CARLA Version 0.9.7 (compiled version)
 - Python version 3.6.9
 - master branch

fabianoboril, 2020-02-05:
The problem is an API change introduced after 0.9.7. Cloudyness became cloudiness. Can you for the moment change this locally at your end?

ravishk1, 2020-02-20:
**Error :- ** AttributeError: 'Namespace' object has no attribute 'config_file'

python3 scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld
Traceback (most recent call last):
  File "scenario_runner.py", line 620, in <module>
    sys.exit(main())
  File "scenario_runner.py", line 610, in main
    result = scenario_runner.run()
  File "scenario_runner.py", line 530, in run
    result = self._run_scenarios()
  File "scenario_runner.py", line 423, in _run_scenarios
    self._args.config_file)
AttributeError: 'Namespace' object has no attribute 'config_file'

closed 2020-03-04 by fabianoboril