carla-simulator/scenario_runner #449: No module named carla error

raeed17, 2020-02-06:
I followed all the instructions in the getting started doc and keep getting the no module name carla error when carla is imported in scenario_runner.py. Any suggestions?

**Desktop):**
 - OS: [Windows 10]
 - CARLA Version [e.g. CARLA 0.9.7]
 - Python version [3.7]

raeed17, 2020-02-06:
C:\Program Files\CARLA_0.9.5\scenario_runner-master>set CARLA_ROOT=C:/'Program Files'/CARLA_0.9.5

C:\Program Files\CARLA_0.9.5\scenario_runner-master>set PYTHONPATH=$PYTHONPATH:C:/'Program Files'/CARLA_0.9.5/PythonAPI/carla/dist/carla-0.9.5-py3.7-win-amd64.egg:C:/'Program Files'/CARLA_0.9.5/PythonAPI/carla/agents:C:/'Program Files'/CARLA_0.9.5/PythonAPI/carla

C:\Program Files\CARLA_0.9.5\scenario_runner-master>python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld
Traceback (most recent call last):
  File "scenario_runner.py", line 30, in <module>
    import carla
ModuleNotFoundError: No module named 'carla'

Here's what I did and the error

kbu9299, 2020-03-31:
Please check your python version: `python --version` and make sure if it's `3.7`

glopezdiest, 2020-06-26:
Hey @raeed17. Did you set up the environment variables? You can check #580 or #367 on how to do so (we are also gonna change the Docs soon to explain)

LikhitaNukala, 2020-06-29:
Hey @glopezdiest , I have executed the instructions given in the docs and updated the environment variables manually. Still, I'm getting the error "No module named Carla". Python version is 3.7.6. Can you please help me with this?

glopezdiest, 2020-06-29:
>  I have executed the instructions given in the docs and updated the environment variables manually

We just updated the Docs, so are you referring to "new" docs or the old ones? In any case, that error is because the CARLA *.egg* isn't being found. Try writing _env_ at a command window an check the PYTHONPATH variable. Is it pointing towards your CARLA *.egg*?

LikhitaNukala, 2020-06-30:
> We just updated the Docs, so are you referring to "new" docs or the old ones? In any case, that error is because the CARLA _.egg_ isn't being found. Try writing _env_ at a command window an check [...]

I edited the environment variables from the Control panel. Anyways, No module named "Carla" error is gone now. Thanks for that. :)
When I execute **python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld**
I'm getting the following error,
Traceback (most recent call last):
  File "scenario_runner.py", line 567, in <module>
    sys.exit(main())
  File "scenario_runner.py", line 556, in main
    scenario_runner = ScenarioRunner(arguments)
  File "scenario_runner.py", line 128, in __init__
    signal.signal(signal.SIGHUP, self._signal_handler)
AttributeError: module 'signal' has no attribute 'SIGHUP'

fabianoboril, 2020-06-30:
@LikhitaNukala  SIGHUP is not available on Windows. We updated SR recently to avoid this problem. Can you update to the latest version?

glopezdiest, 2020-06-30:
@LikhitaNukala As @fabianoboril said, we recently fixed that problem. This is the PR, #561, in case you can't update to the latest version

LikhitaNukala, 2020-06-30:
@fabianoboril I have a scenario runner version 0.9.9, which I have downloaded yesterday. Still I'm getting the error.
So, I have changed the code as @glopezdiest said.

glopezdiest, 2020-06-30:
@LikhitaNukala Yeah, that fix was done very recently and ScenarioRunner 0.9.9 doesn't have it. I'd still recommend using the latest version instead of 0.9.9, as we've made some refactor changes but that's obviously up to you. Closing this issue as it seems to be resolved

closed 2020-06-30 by glopezdiest

SiNa-Dehbari, 2020-11-04:
Hello guys,
I have the same issue.

I've set parameters as below:

set CARLA_ROOT=D:\Softwares\Engineering\Simulators\Carla\WindowsNoEditor
set SCENARIO_RUNNER_ROOT=D:\Softwares\Engineering\Simulators\Carla\scenario_runner-0.9.10
set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI\carla\dist\carla-.egg
set PYTHONPATH=%CARLA_ROOT%\PythonAPI\carla\agents
set PYTHONPATH=%CARLA_ROOT%\PythonAPI\carla
set PYTHONPATH=%CARLA_ROOT%\PythonAPI

after running this code :

python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld

I get this error:

  File "scenario_runner.py", line 44, in <module>
    from srunner.scenarioconfigs.openscenario_configuration import OpenScenarioConfiguration
  File "D:\Softwares\Engineering\Simulators\Carla\scenario_runner-0.9.10\srunner\scenarioconfigs\openscenario_configuration.py", line 21, in <module>
    from srunner.scenarioconfigs.scenario_configuration import ActorConfigurationData, ScenarioConfiguration
  File "D:\Softwares\Engineering\Simulators\Carla\scenario_runner-0.9.10\srunner\scenarioconfigs\scenario_configuration.py", line 65, in <module>
    class ScenarioConfiguration(object):
  File "D:\Softwares\Engineering\Simulators\Carla\scenario_runner-0.9.10\srunner\scenarioconfigs\scenario_configuration.py", line 83, in ScenarioConfiguration
    weather = carla.WeatherParameters()
AttributeError: module 'carla' has no attribute 'WeatherParameters'

I am using:
Window 10,
Python 3.7.6
Scenario Runner 0.9.10

I must use this code for my master project.

Thank you all.

glopezdiest, 2020-11-04:
@SiNa-Dehbari This is related to python not finding the CARLA egg. Could you check that the CARLA *.egg* is at the PYTHONPATH and that the path is correct)

SiNa-Dehbari, 2020-11-04:
Thank you @glopezdiest 
I changed the path and fortunately that error was disappeared.
but still I am not able to run any scenario.

I follow these steps:
1- open CarlaUE4.exe
2- run an example from PythonAPI folder. (to make sure everything works fine)
3- open new terminal on scenario runner folder and run this line of code:

python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld

after this step, my Carla window jumps to new scene and weather changes to sunny or rainy. 
[image]

but still there is no actions for a scenario and after a while I get this error (in the terminal of Scenario runner)

Preparing scenario: FollowLeadingVehicle_1
ScenarioManager: Running scenario FollowVehicle
Not all scenario tests were successful
Please run with --output for further information
Destroying ego vehicle 667
ERROR: failed to destroy actor 667 : unable to destroy actor: not found
No more scenarios .... Exiting

could you please let me know what should I do?
Thank you in advance

glopezdiest, 2020-11-05:
Yeah, you are just missing one more step. This command ```python scenario_runner.py --scenario FollowLeadingVehicle_1 --reloadWorld``` creates the scenario, now you have to launch the manual control (the ScenarioRunner version!!), in order to manually move the ego vehicle.

SiNa-Dehbari, 2020-11-05:
Thank you so much @glopezdiest 
I think I have set the variables permanently on windows.
(when I open a new terminal I don't set variables by command any more and I get no error)

but in the 3rd step, I open new terminal and write this command:

python manual_control.py

and get this error:

Traceback (most recent call last):
  File "manual_control.py", line 67, in <module>
    from examples.manual_control import (World, HUD, KeyboardControl, CameraManager,
ModuleNotFoundError: No module named 'examples'

Have I missed any steps? or is it still related to variable?

and one more thing:
as you are developer of this package and I have to work whit this in future. may I have your email address please?
mine is: dehbari.sina@gmail.com

SiNa-Dehbari, 2020-11-10:
Dear All,
Regarding to my Problem:

`<Traceback (most recent call last):
File "manual_control.py", line 67, in
from examples.manual_control import (World, HUD, KeyboardControl, CameraManager,
ModuleNotFoundError: No module named 'examples'>`

I found that on the tutorial for  setting the variables, on the last line instead of   "   ;   "  is  " : ".

`<set PYTHONPATH=%PYTHONPATH%:%CARLA_ROOT%\PythonAPI>`

and I changed it to ;

`<set PYTHONPATH=%PYTHONPATH%;%CARLA_ROOT%\PythonAPI>`

and now it works.

erouirior, 2021-01-22:
Hi SiNa-Dehbari,

I meet the same error of manual control as well, could you please tell me if I need to set the environment variables in the new terminal for manual control? (FYI, I set the environment variables through system. Sadly, I still meet the error after set the environment variables.) 

Thanks,
Yang

excep, 2022-11-10:
You can also do this : 

go to for ex in my case : CARLA_0.9.13\WindowsNoEditor\PythonAPI\carla\dist 
You'll find an .egg file for example in my case with python version (py-3.7) 

now if you ever wanted to run a python script run it with py -(version of python corresponding to .egg file) 

if you have the version of python on your computer then the code will run. if not a message saying that the python version is not installed. Then you have to install the version of python corresponding to .egg file. and always run python code using : py -"version of corresponding python" example.py

Example : py **-3.7** .\dynamic_weather.py
-3.7 is the version corresponding to .egg file ( my .egg file is : carla-0.9.13-**py3.7**-win-amd64.egg )