carla-simulator/scenario_runner #450: ImportError: No module named 'scene_layout'

aroongta, 2020-02-07:
Traceback (most recent call last):
  File "scenario_runner.py", line 40, in <module>
    from srunner.challenge.autoagents.agent_wrapper import SensorConfigurationInvalid
  File "/home/gauss/Carla_096/scenario_runner/srunner/challenge/autoagents/agent_wrapper.py", line 18, in <module>
    from srunner.challenge.envs.scene_layout_sensors import SceneLayoutReader, ObjectFinder
  File "/home/gauss/Carla_096/scenario_runner/srunner/challenge/envs/scene_layout_sensors.py", line 6, in <module>
    import scene_layout as scene_layout_parser  # This should come from CARLA path
ImportError: No module named 'scene_layout'

**Desktop (please complete the following information):**
 - OS: [ Ubuntu 16.04]
 - CARLA Version [ CARLA 0.9.6]
 - Python version [3.5]
 - Version [ 0.9.6]

fabianoboril, 2020-02-10:
Hi,
please set your PYTHONPATH according to the documentation:
https://github.com/carla-simulator/scenario_runner/blob/master/Docs/getting_started.md

aroongta, 2020-02-14:
Hi,
Thank you for getting back to me. I missed out something while setting the PythonPath initially it seems to be working fine now.
Thank you

closed 2020-02-14 by aroongta