carla-simulator/scenario_runner #455: How do End conditions work in OpenSCENARIO?

amstrudy, 2020-02-12:
I am trying to use the OpenSCENARIO format, but am struggling to get the scenario to continue running. Currently I just have the ego vehicle start up on autopilot and I want it to continue indefinitely, so I put in an end condition for the maneuver that starts the autopilot:

    <End>
        <ConditionGroup>
            <Condition name="EndCondition" delay="0" edge="rising">
                <ByValue>
                    <SimulationTime value="10000" rule="equal_to" />
                </ByValue>
            </Condition>
        </ConditionGroup>
    </End>

But the scenario still just finishes with "Success!" immediately after starting the autopilot. 

What am I doing wrong here? Do the Story Acts just finish as soon as their Sequences have finished executing?

 - OS: Ubuntu 18.04
 - CARLA Version: CARLA 0.9.6
 - Python version: 3.6.9
 - Version: 0.9.6

amstrudy, 2020-02-12:
One fix that works is making another event in the maneuver that just stops the autopilot after a certain number of seconds, but this doesn't really address the question of what the end conditions really do.

fabianoboril, 2020-02-21:
Hi,
the end condition is for maneuvers that would run for an indefinite amount of time (or longer than a certain value) to terminate these maneuvers. The autopilot is however just an on/off, so an immediate action. If you would use a longitudinal action (for example to drive with 10m/s) the endcondition can be used to terminate this maneuver.

closed 2020-02-25 by amstrudy